sn4k3/UVtools #97: [Feature/Change Request] Convert directly to Photon

ComteNiark, 2020-11-13:
Hey, could it be possible to change the way the convert menu works so .photon isn't filed under Chitubox anymore? When you have to process a lot of files, it's annoying to have to change the exact format when selecting the location to save. Chitubox DLP and ZIP can stay as one menu option since they are for use in that piece of software, but .photon files can (and should?) be sent directly to an Anycubic printer.

(Oh and if you ever have the time, please add a dark mode, or at least a partial one! UVTools is a flashbang when I open it!)

sn4k3, 2020-11-13:
I will see if i can split out the extensions as single items.

There's a hidden dark mode theme in UVtools, but not finished, and the big problem, the icons! The current icons are most dark grey, mean you wont see them on dark mode
[image]

sn4k3, 2020-11-14:
Implemented

closed 2020-11-14 by sn4k3

ComteNiark, 2020-11-14:
Thanks for the change!
Honestly this dark mode looks fine to me. Could be improved, sure, but perfectly usable.
I use [Penumbra 10](https://www.deviantart.com/scope10/art/Penumbra-10-Windows-10-visual-style-568740374) and previous versions used to look pretty much like this, but a recent update broke/fixed my windows skin interfering with the interface.

sn4k3, 2020-11-14:
I use native Windows Dark mode. Themes are mostly ok but sometimes they broke and i'm not sure about compability or performance, so i prefer stay on stock for trouble-free.