sn4k3/UVtools #98: [FEATURE REQUEST] Support for Wanhao .CWS

jorgerobles, 2020-11-14:
## Is your feature request related to a problem? Please describe
Just expected to check the Wanhao CWS file, and found that Nova Maker (don't know if earlier or later) has the same extension, but cannot be loaded by UVTOOLS

## Describe the solution you'd like
Wanhao and NovaMaker CWS are almost the same file format, being Wanhao file a XML (.slicing) and NovaMaker a .conf. 
Almost same variable names, and same file structure.  I have no C# proficency but should be easy to detect or just try to get the .slicing after the .conf has failed to load. 

## Describe alternatives you've considered
None for the moment.

## Screenshots
If applicable, add screenshots to help explain your request.

## Additional context

```
<?xml version="1.0" encoding="utf-8"?>
<SliceBuildConfig FileVersion="2">
  <DotsPermmX>21.1640211640212</DotsPermmX>
  <DotsPermmY>21.1640211640212</DotsPermmY>
  <XResolution>2560</XResolution>
  <YResolution>1440</YResolution>
  <BlankTime>0</BlankTime>
  <TopTime>0</TopTime>
  <PlatformTemp>75</PlatformTemp>
  <ExportSVG>0</ExportSVG>
  <Export>True</Export>
  <ExportPNG>True</ExportPNG>
  <CalcSliceSize>False</CalcSliceSize>
  <XOffset>0</XOffset>
  <YOffset>0</YOffset>
  <Direction>Bottom_Up</Direction>
  <LiftDistance>5</LiftDistance>
  <SlideTiltValue>0</SlideTiltValue>
  <SettleTime>1600</SettleTime>
  <AntiAliasing>False</AntiAliasing>
  <UseMainLiftGCode>False</UseMainLiftGCode>
  <AntiAliasingValue>1</AntiAliasingValue>
  <LiftFeedRate>100</LiftFeedRate>
  <BottomLiftFeedRate>90</BottomLiftFeedRate>
  <LiftRetractRate>150</LiftRetractRate>
  <ExportOption>ZIP</ExportOption>
  <RenderOutlines>False</RenderOutlines>
  <OutlineWidth_Inset>2</OutlineWidth_Inset>
  <OutlineWidth_Outset>0</OutlineWidth_Outset>
  <FlipX>False</FlipX>
  <FlipY>False</FlipY>
  <Notes>Use these as a place to start; tune them for your specific printer and resins.
Always run a calibration on your resin to dial in for your machine.
Profiles created by Earl Miller for CW v1.0.0.75.
</Notes>
  <GCodeHeader>;********** Header Start ********
G21 ; Set units to be mm
G91 ; Relative Positioning
M17 ; Enable motors
; G28 Z ; Auto-home on print - commented by default
</GCodeHeader>
  <GCodeFooter>;********** Footer ********
@cmdCloseShutter ; UV off
G4 P0            ; wait for last lift to complete
G1 Z40.0 F150.0  ; lift model clear of resin
G4 P0            ; sync
M18              ; Disable Motors
;&lt;Completed&gt;
</GCodeFooter>
  <GCodePreslice>;********** Pre-Slice %d$CURSLICE ********
G4 P0            ; Make sure any previous relative moves are complete
;&lt;Delay&gt; %d$SettleTime
</GCodePreslice>
  <GCodeLift>;********** Lift Sequence %d$CURSLICE ********
[... 199 lines cut ...]
  </InkConfig>
  <InkConfig>
    <Name>Magma TR Red 35um</Name>
    <SliceHeight>0.0350000001490116</SliceHeight>
    <LayerTime>7500</LayerTime>
    <FirstLayerTime>40000</FirstLayerTime>
    <NumberofBottomLayers>5</NumberofBottomLayers>
    <ResinPriceL>0</ResinPriceL>
  </InkConfig>
  <InkConfig>
    <Name>elegoo</Name>
    <SliceHeight>0.0350000001490116</SliceHeight>
    <LayerTime>8000</LayerTime>
    <FirstLayerTime>70000</FirstLayerTime>
    <NumberofBottomLayers>8</NumberofBottomLayers>
    <ResinPriceL>0</ResinPriceL>
  </InkConfig>
  <MinTestExposure>5000</MinTestExposure>
  <TestExposureStep>1000</TestExposureStep>
  <ExportPreview>None</ExportPreview>
  <UserParameters />
  <UseMask>False</UseMask>
  <MaskFilename>C:\EDITEDSOMEPATH\mask.png</MaskFilename>
</SliceBuildConfig>
```

sn4k3, 2020-11-14:
What slicer you use? What printer model?

jorgerobles, 2020-11-15:
Printer: Wanhao d7, 
Slicers: Creation Workshop CWX 1.2.4 and Wanhao Creation Workshop (lychee brother)

sn4k3, 2020-11-15:
What a pain just go find that slicers and deal with them...

jorgerobles, 2020-11-15:
Tested output files both CWX 1.2.4 and Wanhao Workshop.
Exported to CWS to be used with Wanhao d7 box (bananapi)

CWX exported:
* {profile_name}.slicing (similar to Novamaker .conf but in XML)
* manifest.xml, containing a list of the PNG files
* {file_name}.gcode
* bunch of pngs

Wanhao Creation Workshop (WCW - Mango.io):
* manifest.xml, containing a list of the PNG files
* {file_name}.gcode
* bunch of pngs

manifest.xml
```
<?xml version="1.0" encoding="UTF-8"?>
<manifest FileVersion="1">
  <Slices>
    <Slice>
      <name>mango3d0000.png</name>
    </Slice>
 ...
    <Slice>
      <name>mango3d0198.png</name>
    </Slice>
    <Slice>
      <name>mango3d0199.png</name>
    </Slice>
  </Slices>
  <GCode>
    <name>mango3d.gcode</name>
  </GCode>
</manifest>
```

My findings are that XML manifest .gcode and .PNGs are needed for printing with Wanhao Box. 
(My experience with Box have not been good, some hangs, so rather print using oldie CWX, as the printing computer feels better with it rather than resource consuming WCW)

But .slicing includes resin speeds and parameters to rebuild the GCODE, and might be needed if you want to supporting full compatibility with native slicers/conversions (not my case, only want to do pixel operations).

Thanks in advance,
Jorge.

sn4k3, 2020-11-19:
Implemented

closed 2020-11-19 by sn4k3