sn4k3/UVtools #100: [BUG] Move directions are nonintuitive

nvx, 2020-11-15:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
When dealing with a .photon file, doing a "move" operation 200 pixels "left" results in the image being moved down in the preview.

I assume this is because the file format is actually portrait, but for convenience the layers are normally shown in landscape view - but from a UX point of view it's rather confusing.

## Expected behavior
Operation directions should match the orientation in the preview, or maybe better terminology than left/right/up/down (X-Y and mention how wide each dimension is?)

## Additional context
I imagine this also affects other areas where directions are mentioned too (eg, the feature added for #94) but I can't easily confirm that.

sn4k3, 2020-11-15:
Operations are always done over original images.
With rotation enabled that is just a virtual preview and is not linked.
When performing such operations like move, pattern and others you should disable "Rotate" preview

nvx, 2020-11-16:
That makes sense. Maybe just adding a note saying as such would save some confusion. I must admit I wasted longer than I'd like to admit trying to figure out why it wasn't working how I expected it to the first time.

sn4k3, 2020-11-16:
Yes i can put a note on operations to renember the user

sn4k3, 2020-11-19:
Note implemented

closed 2020-11-19 by sn4k3