sn4k3/UVtools #99: [BUG]  Occasional print XY movement observed in chitubox after fixing up in UVTools

T1ckL35, 2020-11-15:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
I have imported a chitubox sliced .ctb file for the Elegoo Saturn into UVTools 1.3.0 and fixed it up. I run the Issue detection (removing reservoir detection) and then run a repair. After that I use the pixel editor to fix up a couple of issues and apply it. Finally I save that file to a new named one. I then double check the file by reloading into chitubox and occasionally have noticed that the whole print area has been offset. It seems that sometimes UVTools may have modified the print so that the left or right edge of the detected area has been recalculated as the center.

## To Reproduce
Steps to reproduce the behavior:
(Note, this seems to happen sometimes but not always)
1. Import a chitubox sliced .ctb file for the Elegoo Saturn into UVTools 1.3.0
2. Run Issue detection (not with reservoir detection enabled)
3. Run repair
4. Use pixel editor to fix up some issues and apply the changes
5. Save the file as a new one
6. Reload the file in chitubox to check it looks ok

## Expected behavior
The new fixed file should always have the print in the same location on the build plate

## Screenshots
Screenshot showing the occasional offset issue: https://ibb.co/h7s4BPs
Screenshot showing the expected positioning: https://ibb.co/6mHLHrq

## System
 - OS: [Windows 10 x64]
- Software: UVTools 1.3.0 (also seen on the last 1.2 release)

nvx, 2020-11-15:
Weird, what happens if you reopen the affected file in UV Tools again? Does it show as offset there too or?

T1ckL35, 2020-11-15:
Hi, I annoyingly may have deleted the files but will save any new ones that exhibit the behaviour and will reopen them in UVTools to see.

sn4k3, 2020-11-15:
Repairs don't move anything, it can be a bug with encoding, properties or slicer. 
When you able to replicate the problem, please attach the problematic file here, the original and the modified

T1ckL35, 2020-11-15:
Well I've just managed to reproduce it in this way:
- Open chitubox, slice and save. Leave chituboc open in the background
- Open UVTools, anaylse, repair and fixup. Then save a new file from it
- In Chitubox open the new file (note, it hasn't been shut). Occasionally it shows the original offset.

After this I have just tried:
- Shutdown Chitubox
- Open file in Chitubox, no problem displayed
- Open file in UVTools, no problem displayed.

So it looks like it is a slicer issue rather than anything done in UVTools.
Apologies for thinking it may be UVTools related.

sn4k3, 2020-11-15:
If file still open on UVtools without an offset most likely is a slicer problem. If you are saving as new file i can't see why chitubox mess up

closed 2020-11-17 by sn4k3