sn4k3/UVtools #101: Adjust layer height [BUG]

rahsoft, 2020-11-19:
All including version 1.3.2

If file layer resolution is 0.01 or 0.015 adjust layer height is not possible. Please let to convert 0.01=>0.02 and 0.015=>0.03.
Thanks for tool - it has big future.

sn4k3, 2020-11-19:
UVtools only allow layer re-height when both height and layer count math match without fractions, you can't have half layers, and if i allow that you will loose quality. So it's a safety feature to ensure model comes out ok. Thats why UVtools give you a allowed list instead of a user manual input.
BTW never use 0.00x you are losing Z precision, most printers are 0.01mm at best.

Tell me your LayerCount at 0.01 so i can confirm the math is ok

rahsoft, 2020-11-19:
> Tell me your LayerCount at 0.01mm so i can confirm the math is ok

Hello,

 I am trying to slice at 0.01 (or 0.015) do some manipulations with layers
and want to output at 0.02 (or 0.03).

P.S. Manipulations - rasterization of layers...

Sincerely Yours,
Renatas

sn4k3, 2020-11-19:
How many layers the file have? 
And why not slice at correct layer height?

rahsoft, 2020-11-19:
I use Your app to rasterize layers and then combine them - some kind of cheap 3D antialiassing (on Z axis)...

sn4k3, 2020-11-26:
Can't have half layers nor dynamic layer heights, UVtools limits Z precision to 0.01mm and 2 decimal plates. This is all safety features. Override this will do bad prints somehow.

To know the rules behind you can read this: https://github.com/sn4k3/UVtools/blob/master/UVtools.Core/Operations/OperationLayerReHeight.cs#L66
In other words: Layer count and layer height divided by a factor must be both integers, if decimals it will discard that factor

rahsoft, 2020-11-30:
I wonder that maybe first break in loop breaks out from procedure...
Maybe this will make it work with 0.01 to 0.02, 0.03, ...

public static OperationLayerReHeightItem[] GetItems(uint layerCount, decimal layerHeight)
        {
            List<OperationLayerReHeightItem> list = new List<OperationLayerReHeightItem>();
            for (byte i = 2; i < 255; i++) // Lower
            {
                if (layerHeight / i < 0.01m) **continue**;
....

sn4k3, 2020-11-30:
It won't. I already explaned why it won't work. It depends on your layer number.

rahsoft, 2020-11-30:
I dont want to divide - I want multiply.
0.01 input - x2 = 0.02, x3 = 0.03 outputs.

P.S. Layer count is not problem - tryed dividing by 2, 3, 4 - same error...

sn4k3, 2020-11-30:
The problem is the same no matter divide or multiply.
I ask you before what was your layer count and layer height, never got a response...

rahsoft, 2020-12-01:
Becouse I tryed all combinations without luck - clonnig layers 1 by 1 ...

P.S. So there is no way it can be done ? No hope for future versions ?

sn4k3, 2020-12-01:
You keep not anwsering to my question, how many layers your file have and the height?
If you hide the only valid information for the problem how can anyone help you?

rahsoft, 2020-12-01:
For test it was 328 layers at 0.01.
Output from Chitubox 1.7.0 to .photon file. Antialliasing x8.

rahsoft, 2020-12-01:
Test file
[https://drive.google.com/file/d/1BG9w98uSi-xLWiC1vpHn8vLAaGYQHc9W/view?usp=sharing](url)

Same file sliced at 0.02 (164 layers) works (to 0.04, 0.06)...

sn4k3, 2020-12-01:
Ok found the problem, it was a rounding problem that won't happen on every case. Will fix on next patch

sn4k3, 2020-12-04:
Fixed

closed 2020-12-04 by sn4k3

rahsoft, 2020-12-04:
Thank You.
Confirm - it's working.