sn4k3/UVtools #102: [BUG] Wanhao Workshop CWS cannot be loaded

jorgerobles, 2020-11-21:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
Cannot load Wanhao Workshop File

## To Reproduce
Steps to reproduce the behavior:
Open attached file

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots
If applicable, add screenshots to help explain your problem. Never hide information, full GUI/window screenshot is always a must!

## Files
Were possible attach the problematic file, both source and output file if applicable.
[eiffelTower_d7w.cws.zip](https://github.com/sn4k3/UVtools/files/5578216/eiffelTower_d7w.cws.zip)

## System
**Desktop (please complete the following information):**
 - OS: Mac Os X Catalina 10.15.4, Imac 2015
  
## Exceptions Log file
In some unexpected exceptions errors will generate and save a log file under appfolder/Logs/AppErrors.
If present you can attach the log file here. Check file name against date/time to be sure it's related to your problem.

```
System.ArgumentException: [[totalSlices]] is not a valid value for UInt32. (Parameter 'value')
 ---> System.FormatException: Input string was not in a correct format.
   at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type)
   at System.Number.ParseUInt32(ReadOnlySpan`1 value, NumberStyles styles, NumberFormatInfo info)
   at System.UInt32.Parse(String s, NumberStyles style, IFormatProvider provider)
   at System.ComponentModel.UInt32Converter.FromString(String value, NumberFormatInfo formatInfo)
   at System.ComponentModel.BaseNumberConverter.ConvertFrom(ITypeDescriptorContext context, CultureInfo culture, Object value)
   --- End of inner exception stack trace ---
   at System.ComponentModel.BaseNumberConverter.ConvertFrom(ITypeDescriptorContext context, CultureInfo culture, Object value)
   at System.ComponentModel.TypeConverter.ConvertFromString(String text)
   at UVtools.Core.Extensions.StringExtensions.Convert[T](String input) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Extensions\StringExtensions.cs:line 53
   at UVtools.Core.Helpers.SetPropertyValue(PropertyInfo attribute, Object obj, String value) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Helpers.cs:line 85
   at UVtools.Core.FileFormats.CWSFile.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\CWSFile.cs:line 749
   at UVtools.WPF.MainWindow.<>c__DisplayClass91_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 824
```

## Additional context
Add any other context about the problem here.

.gcode File shows a kind of pragma [[totalSlices]] that causes strong typed uint32 property to fail. Supposed to be Manifest.xml slice count.
Cannot assure if it's a needed parameter, so maybe bypassing exception (casting to 0) should be enough to load the file and do standard operations. (Event doing with each of the properties failing casting and just pop an warning or log could be helpy)

[image]

sn4k3, 2020-11-21:
I based the gcode parser on CWX. It seen diferent slicers use diferent gcode for the printer, god damit... Also number of slices = number of layers, no sense redundancy
I will fix that on next release.

jorgerobles, 2020-11-21:
Seems got WCW misconfigured, sorry. Here´s a proper file.
[mango3d2.gcode.zip](https://github.com/sn4k3/UVtools/files/5578371/mango3d2.gcode.zip)

> My big advice here is use PrusaSlicer instead and convert to your printer... even when i fix that, UVtools will have missing data with the information your slicer is outputing.

Will try, thanks!

Really printer vendors should adhere an standard 🙄  
Luckily, We've got UVTOOLS  😄

sn4k3, 2020-11-21:
I just confirmed, things like resolution, layer height and other informations required by uvtools are missing on that slicer format. Is possible to get them via calculations but i think not worth to have all that extra conditions for such format. CWX otherwise is full compatible.
You can setup a new printer on prusaslicer and work from there

sn4k3, 2020-11-25:
fixed

closed 2020-11-25 by sn4k3