sn4k3/UVtools #103: [FEATURE REQUEST] Recalculate print time

nvx, 2020-11-25:
## Is your feature request related to a problem? Please describe
At the moment when you adjust print parameters (especially important if slicing in eg PrusaSlicer then converting to eg Photon, as not all options in the Photon file format can be set in PrusaSlicer, so I can't really avoid adjusting print parameters in my workflow) doesn't recalculate the print time. Even when using Chitubox to slice for eg a Photon, Chitubox seems to have pretty inaccurate print time calculations anyway.

## Describe the solution you'd like
Having UV Tools recalculate the estimated print time, either automatically when editing print parameters that affect the print time (lift speeds/heights, exposure times, etc), or on demand would be super handy.

Having a dynamically updating print time estimate available on the edit print parameters window would be super handy so you can see just how much time you're trading off when tweaking parameters like lift speeds/heights too.

Using the same logic as the light off delay calculator would be handy too to see when setting a lift height+speed combination that ends up being longer than your configured light off delay - I suspect this is where the errors in chitubox's estimates come from by not taking this into account.

sn4k3, 2020-11-26:
Implemented.
The diff on edit parameters will be harder since the avaliable parameters are dynamic and file dependent... Still you can apply settings and look at bottom bar to cath the diference.
Please confirm time accuracy with a real print.

closed 2020-11-26 by sn4k3

nvx, 2021-03-13:
I finally got around to setting up my printers ethernet port talking to Home Assistant so I can get alerts when prints finish. The other side effect of this is I can now tell to within a minute how long prints take in practice (I only poll the printer status once a minute go figure).

I'm glad to say that the calculated time in UVTools and the actual time it took to print match exactly for my OG Photon (6h 47m print).

sn4k3, 2021-03-13:
Good to know, it will only fail at printers that overrides the settings or somehow change them, then uvtools calculation is no longer real