sn4k3/UVtools #104: [QUESTION] Run on manjaro/arch

torels, 2020-11-26:
Hello,

How do I get to run this software on manjaro linux?
I seem to not be able to find the packages for packages-microsoft-prod on microsoft's website, and debtap won't do the trick.
I installed mono on my machine, and when I try to run UVtools it actually runs, but it bumps in a dependecy problem stating that
```
UVtools can not run due lack of dependencies from cvextern/OpenCV
Please build or install this dependencies in order to run UVtools
Check manual or page at 'Requirements' section for help

Additional information:
System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'cvextern' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libcvextern: cannot open shared object file: No such file or directory
   at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 73
```

Is there any way I can work around this manually?

sn4k3, 2020-11-26:
dotnet is no longer required but other dependecies are: https://github.com/sn4k3/UVtools#linux
You need find a way to install them, try using the arch solution to install packages and use same packages names.
I can install a VM latter and try.

As a extreme solution you can compile every dependency, but its a slow process...

sn4k3, 2020-11-26:
I have tried with a VM, but theres a problem with dependencies, even if i install them, they show as missing.
You can try on your machine: sudo pacman -S openjpeg2 libjpeg-turbo libpng libgeotiff libdc1394 libdc1394 ffmpeg openexr tbb dotnet-runtime

cubiq, 2020-11-27:
the two libraries are actually available, but in an older version. You may create a symlink to them, but even so I get an error:

```dotnet: symbol lookup error: libcvextern.so: undefined symbol: _ZN7Imf_2_314ChromaticitiesC1ERKN9Imath_2_34Vec2IfEES5_S5_S5_```

ldd'ing libcvextern.so it says

```./libcvextern.so: /usr/lib/libtiff.so.5: no version information available (required by ./libcvextern.so)```

that might be some version mismatch

ampotos, 2020-12-09:
I have the same issue and I think the issue is with libIlmImf as my system has 2.5 instead of 2.3 installed.
Would it be possible to include the correct version of the library in the realease zip ?

sn4k3, 2020-12-09:
I'm trying to compile them but my manjaro vm acts stange and turn black when i try compile sources. Will try on pure Arch latter

sn4k3, 2020-12-27:
Hi, can you try to manual compile the dependencies?

```bash
sudo pacman -S base-devel git cmake
git clone https://github.com/emgucv/emgucv emgucv 
cd emgucv
git submodule update --init --recursive
cd platforms/ubuntu/20.04
./cmake_configure
cmake build
cd build; make; cd ..
```

ominds, 2020-12-28:
Tried manual compiling (as user, not root). It didn't work. To be clear, here's what I first got

```
The framework 'Microsoft.NETCore.App', version '5.0.0' was not found.
  - The following frameworks were found:
      3.1.8 at [/usr/share/dotnet/shared/Microsoft.NETCore.App]
      5.0.0-rc.2.20475.5 at [/usr/share/dotnet/shared/Microsoft.NETCore.App]
```
I changed UVtools.runtimeconfig.json version to "5.0.0-rc.2.20475.5"

**Then I got this**

[image]

I'm running archlinux.

> Hi, can you try to manual compile the dependencies?
> 
> ```shell
> sudo pacman -S base-devel git cmake
> git clone https://github.com/emgucv/emgucv emgucv 
> cd emgucv
> git submodule update --init --recursive
> cd platforms/ubuntu/20.04
> ./cmake_configure
> cmake build
> cd build; make; cd ..
> ```
> 
> My VM just turn back if i try :|

sn4k3, 2020-12-28:
Dont compile UVtools, you need the dependencies, follow the previous build commands i sent for EmguCV -- This is not UVtools

ominds, 2020-12-28:
I understand. I did the exact commands you sent on EmguCV, except I didn't perform the last 3 as root.

I didn't compile UVtools. I was just showing what I got when I tried running it after installing EmguCV.

I'll try the last 3 commands as root and hope that gives a different result and will report back.

ominds, 2020-12-28:
OK. I compiled EmguCV and ran last 3 commands as root. When I run UVtools.sh here's what I get

```
sh UVtools.sh 
It was not possible to find any compatible framework version
The framework 'Microsoft.NETCore.App', version '5.0.0' was not found.
  - The following frameworks were found:
      3.1.8 at [/usr/share/dotnet/shared/Microsoft.NETCore.App]
      5.0.0-rc.2.20475.5 at [/usr/share/dotnet/shared/Microsoft.NETCore.App]

You can resolve the problem by installing the specified framework and/or SDK.

The specified framework can be found at:
  - https://aka.ms/dotnet-core-applaunch?framework=Microsoft.NETCore.App&framework_version=5.0.0&arch=x64&rid=arch-x64

```
I've edited UVtools.runtimeconfig.json and changed the Microsoft.NETCore.App to 5.0.0-rc.2.20475.5

And I get this when I run sh UVtools.sh

[image]

So no change

sn4k3, 2020-12-28:
Paste here the output of `ldd libcvextern.so`

ominds, 2020-12-28:
```
ldd: warning: you do not have execution permission for `./libcvextern.so'
./libcvextern.so: /usr/lib/libtiff.so.5: no version information available (required by ./libcvextern.so)
	linux-vdso.so.1 (0x00007ffd255f8000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f8b86bbd000)
	libtiff.so.5 => /usr/lib/libtiff.so.5 (0x00007f8b86b2d000)
	libgeotiff.so.5 => /usr/lib/libgeotiff.so.5 (0x00007f8b86af7000)
	libdc1394.so.22 => not found
	libavcodec.so.58 => /usr/lib/libavcodec.so.58 (0x00007f8b854ed000)
	libavformat.so.58 => /usr/lib/libavformat.so.58 (0x00007f8b85263000)
	libavutil.so.56 => /usr/lib/libavutil.so.56 (0x00007f8b84fb4000)
	libswscale.so.5 => /usr/lib/libswscale.so.5 (0x00007f8b84f25000)
	libgtk-3.so.0 => /usr/lib/libgtk-3.so.0 (0x00007f8b8477d000)
	libgdk-3.so.0 => /usr/lib/libgdk-3.so.0 (0x00007f8b8467e000)
	libcairo.so.2 => /usr/lib/libcairo.so.2 (0x00007f8b8455f000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0 (0x00007f8b8453a000)
	libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0 (0x00007f8b844e0000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f8b843ac000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f8b842d8000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f8b84204000)
	libjpeg.so.8 => /usr/lib/libjpeg.so.8 (0x00007f8b84180000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f8b84149000)
	libIlmImf-2_3.so.24 => not found
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007f8b84101000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f8b840fb000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f8b840d9000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f8b83efc000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f8b83db6000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f8b83d9a000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f8b83bd1000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f8b8933f000)
	libzstd.so.1 => /usr/lib/libzstd.so.1 (0x00007f8b83aff000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f8b83ad7000)
	libproj.so.15 => /usr/lib/libproj.so.15 (0x00007f8b837e8000)
	libswresample.so.3 => /usr/lib/libswresample.so.3 (0x00007f8b837c9000)
	libvpx.so.6 => /usr/lib/libvpx.so.6 (0x00007f8b834c2000)
	libwebpmux.so.3 => /usr/lib/libwebpmux.so.3 (0x00007f8b834b6000)
	libwebp.so.7 => /usr/lib/libwebp.so.7 (0x00007f8b83446000)
	libdav1d.so.4 => /usr/lib/libdav1d.so.4 (0x00007f8b83311000)
	libopencore-amrwb.so.0 => /usr/lib/libopencore-amrwb.so.0 (0x00007f8b832fb000)
	libaom.so.2 => /usr/lib/libaom.so.2 (0x00007f8b82c27000)
	libgsm.so.1 => /usr/lib/libgsm.so.1 (0x00007f8b82c16000)
	libmp3lame.so.0 => /usr/lib/libmp3lame.so.0 (0x00007f8b82b9e000)
	libopencore-amrnb.so.0 => /usr/lib/libopencore-amrnb.so.0 (0x00007f8b82b72000)
	libopenjp2.so.7 => /usr/lib/libopenjp2.so.7 (0x00007f8b82b17000)
	libopus.so.0 => /usr/lib/libopus.so.0 (0x00007f8b82ab9000)
	librav1e.so.0 => /usr/lib/librav1e.so.0 (0x00007f8b828be000)
	libspeex.so.1 => /usr/lib/libspeex.so.1 (0x00007f8b828a1000)
	libtheoraenc.so.1 => /usr/lib/libtheoraenc.so.1 (0x00007f8b82868000)
	libtheoradec.so.1 => /usr/lib/libtheoradec.so.1 (0x00007f8b8284f000)
	libvorbis.so.0 => /usr/lib/libvorbis.so.0 (0x00007f8b82821000)
	libvorbisenc.so.2 => /usr/lib/libvorbisenc.so.2 (0x00007f8b82776000)
	libx264.so.160 => /usr/lib/libx264.so.160 (0x00007f8b824b7000)
	libx265.so.192 => /usr/lib/libx265.so.192 (0x00007f8b81223000)
	libxvidcore.so.4 => /usr/lib/libxvidcore.so.4 (0x00007f8b81114000)
	libva.so.2 => /usr/lib/libva.so.2 (0x00007f8b810e7000)
	libmfx.so.1 => /usr/lib/libmfx.so.1 (0x00007f8b810d9000)
	libxml2.so.2 => /usr/lib/libxml2.so.2 (0x00007f8b80f6b000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f8b80f58000)
	libmodplug.so.1 => /usr/lib/libmodplug.so.1 (0x00007f8b80dc6000)
	libbluray.so.2 => /usr/lib/libbluray.so.2 (0x00007f8b80d76000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x00007f8b80cd5000)
	libgnutls.so.30 => /usr/lib/libgnutls.so.30 (0x00007f8b80ad2000)
	libsrt.so.1 => /usr/lib/libsrt.so.1 (0x00007f8b80a11000)
	libssh.so.4 => /usr/lib/libssh.so.4 (0x00007f8b809a5000)
	libva-drm.so.2 => /usr/lib/libva-drm.so.2 (0x00007f8b8099e000)
	libva-x11.so.2 => /usr/lib/libva-x11.so.2 (0x00007f8b80996000)
	libvdpau.so.1 => /usr/lib/libvdpau.so.1 (0x00007f8b80991000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f8b80850000)
	libdrm.so.2 => /usr/lib/libdrm.so.2 (0x00007f8b8083b000)
[... 38 lines cut ...]
	libp11-kit.so.0 => /usr/lib/libp11-kit.so.0 (0x00007f8b7fc55000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f8b7fc34000)
	libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007f8b7fab0000)
	libtasn1.so.6 => /usr/lib/libtasn1.so.6 (0x00007f8b7fa9a000)
	libnettle.so.8 => /usr/lib/libnettle.so.8 (0x00007f8b7fa5c000)
	libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x00007f8b7fa14000)
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x00007f8b7f736000)
	libmount.so.1 => /usr/lib/libmount.so.1 (0x00007f8b7f6d7000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f8b7f6bd000)
	libthai.so.0 => /usr/lib/libthai.so.0 (0x00007f8b7f6b2000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007f8b7f683000)
	libdbus-1.so.3 => /usr/lib/libdbus-1.so.3 (0x00007f8b7f62e000)
	libatspi.so.0 => /usr/lib/libatspi.so.0 (0x00007f8b7f5f5000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007f8b7f5f0000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007f8b7f5e8000)
	libgomp.so.1 => /usr/lib/libgomp.so.1 (0x00007f8b7f5a7000)
	libicudata.so.68 => /usr/lib/libicudata.so.68 (0x00007f8b7da64000)
	libblkid.so.1 => /usr/lib/libblkid.so.1 (0x00007f8b7da13000)
	libdatrie.so.1 => /usr/lib/libdatrie.so.1 (0x00007f8b7da09000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f8b7d953000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f8b7d930000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f8b7d80d000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f8b7d7e6000)

```

sn4k3, 2020-12-28:
Ok so, there's 2 dependencies missing:

```
libdc1394.so.22 => not found
libIlmImf-2_3.so.24 => not found
```

They should be installed if the compile works. Does it end with error? the configure, cmake or make command?

ominds, 2020-12-28:
I happened to find that same file in emgucv folder I cloned under libs/libcvextern.so (I don't know if this is relevant at all)

I tried to ldd it to see if it gives same output... here's the output

```
	linux-vdso.so.1 (0x00007ffc9bbe6000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f0e039b9000)
	libgeotiff.so.5 => /usr/lib/libgeotiff.so.5 (0x00007f0e03983000)
	libSM.so.6 => /usr/lib/libSM.so.6 (0x00007f0e03979000)
	libICE.so.6 => /usr/lib/libICE.so.6 (0x00007f0e0395c000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f0e0381b000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007f0e03806000)
	libXt.so.6 => /usr/lib/libXt.so.6 (0x00007f0e03799000)
	libOpenGL.so.0 => /usr/lib/libOpenGL.so.0 (0x00007f0e0376d000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007f0e0373a000)
	libGLU.so.1 => /usr/lib/libGLU.so.1 (0x00007f0e036c5000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f0e035f1000)
	libdc1394.so.25 => /usr/lib/libdc1394.so.25 (0x00007f0e03577000)
	libgstbase-1.0.so.0 => /usr/lib/libgstbase-1.0.so.0 (0x00007f0e034f4000)
	libgstreamer-1.0.so.0 => /usr/lib/libgstreamer-1.0.so.0 (0x00007f0e033ab000)
	libgstapp-1.0.so.0 => /usr/lib/libgstapp-1.0.so.0 (0x00007f0e03399000)
	libgstriff-1.0.so.0 => /usr/lib/libgstriff-1.0.so.0 (0x00007f0e03388000)
	libgstpbutils-1.0.so.0 => /usr/lib/libgstpbutils-1.0.so.0 (0x00007f0e03349000)
	libgstaudio-1.0.so.0 => /usr/lib/libgstaudio-1.0.so.0 (0x00007f0e032cc000)
	libgstvideo-1.0.so.0 => /usr/lib/libgstvideo-1.0.so.0 (0x00007f0e03214000)
	libgsttag-1.0.so.0 => /usr/lib/libgsttag-1.0.so.0 (0x00007f0e031d3000)
	libavcodec.so.58 => /usr/lib/libavcodec.so.58 (0x00007f0e01bc9000)
	libavformat.so.58 => /usr/lib/libavformat.so.58 (0x00007f0e0193f000)
	libavutil.so.56 => /usr/lib/libavutil.so.56 (0x00007f0e01692000)
	libswscale.so.5 => /usr/lib/libswscale.so.5 (0x00007f0e01603000)
	libgtk-3.so.0 => /usr/lib/libgtk-3.so.0 (0x00007f0e00e59000)
	libgdk-3.so.0 => /usr/lib/libgdk-3.so.0 (0x00007f0e00d5a000)
	libpangocairo-1.0.so.0 => /usr/lib/libpangocairo-1.0.so.0 (0x00007f0e00d49000)
	libpango-1.0.so.0 => /usr/lib/libpango-1.0.so.0 (0x00007f0e00cf7000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f0e00c23000)
	libatk-1.0.so.0 => /usr/lib/libatk-1.0.so.0 (0x00007f0e00bfa000)
	libcairo-gobject.so.2 => /usr/lib/libcairo-gobject.so.2 (0x00007f0e00bec000)
	libcairo.so.2 => /usr/lib/libcairo.so.2 (0x00007f0e00acd000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0 (0x00007f0e00aa8000)
	libgio-2.0.so.0 => /usr/lib/libgio-2.0.so.0 (0x00007f0e008ec000)
	libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0 (0x00007f0e00894000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f0e00760000)
	libgthread-2.0.so.0 => /usr/lib/libgthread-2.0.so.0 (0x00007f0e00759000)
	libwebp.so.7 => /usr/lib/libwebp.so.7 (0x00007f0e006e9000)
	libopenjp2.so.7 => /usr/lib/libopenjp2.so.7 (0x00007f0e0068e000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f0e0066c000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f0e00666000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f0e0065b000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f0e0047c000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f0e00336000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f0e0031c000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f0e00153000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f0e0905d000)
	libtiff.so.5 => /usr/lib/libtiff.so.5 (0x00007f0e000c3000)
	libproj.so.15 => /usr/lib/libproj.so.15 (0x00007f0dffdd4000)
	libuuid.so.1 => /usr/lib/libuuid.so.1 (0x00007f0dffdc9000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f0dffd9f000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007f0dffce8000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007f0dffc62000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f0dffc4f000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f0dffc16000)
	libraw1394.so.11 => /usr/lib/libraw1394.so.11 (0x00007f0dffc04000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x00007f0dffbe7000)
	libgmodule-2.0.so.0 => /usr/lib/libgmodule-2.0.so.0 (0x00007f0dffbe2000)
	libunwind.so.8 => /usr/lib/libunwind.so.8 (0x00007f0dffbc8000)
	libdw.so.1 => /usr/lib/libdw.so.1 (0x00007f0dffb2a000)
	liborc-0.4.so.0 => /usr/lib/liborc-0.4.so.0 (0x00007f0dffaa3000)
	libswresample.so.3 => /usr/lib/libswresample.so.3 (0x00007f0dffa84000)
	libvpx.so.6 => /usr/lib/libvpx.so.6 (0x00007f0dff77f000)
	libwebpmux.so.3 => /usr/lib/libwebpmux.so.3 (0x00007f0dff773000)
[... 63 lines cut ...]
	libelf.so.1 => /usr/lib/libelf.so.1 (0x00007f0dfc564000)
	libsoxr.so.0 => /usr/lib/libsoxr.so.0 (0x00007f0dfc4e2000)
	libogg.so.0 => /usr/lib/libogg.so.0 (0x00007f0dfc4d5000)
	libmvec.so.1 => /usr/lib/libmvec.so.1 (0x00007f0dfc4a9000)
	libicuuc.so.68 => /usr/lib/libicuuc.so.68 (0x00007f0dfc2ba000)
	libp11-kit.so.0 => /usr/lib/libp11-kit.so.0 (0x00007f0dfc185000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f0dfc164000)
	libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007f0dfbfe0000)
	libtasn1.so.6 => /usr/lib/libtasn1.so.6 (0x00007f0dfbfca000)
	libnettle.so.8 => /usr/lib/libnettle.so.8 (0x00007f0dfbf8c000)
	libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x00007f0dfbf44000)
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x00007f0dfbc66000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007f0dfbc35000)
	libdbus-1.so.3 => /usr/lib/libdbus-1.so.3 (0x00007f0dfbbe0000)
	libatspi.so.0 => /usr/lib/libatspi.so.0 (0x00007f0dfbba9000)
	libdatrie.so.1 => /usr/lib/libdatrie.so.1 (0x00007f0dfbb9f000)
	libblkid.so.1 => /usr/lib/libblkid.so.1 (0x00007f0dfbb4e000)
	libgomp.so.1 => /usr/lib/libgomp.so.1 (0x00007f0dfbb0b000)
	libicudata.so.68 => /usr/lib/libicudata.so.68 (0x00007f0df9fca000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f0df9f14000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f0df9ef1000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f0df9dd0000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f0df9da7000)

```

sn4k3, 2020-12-28:
Ok that seen to have the linked libs. replace that new libcvextern.so with the one in uvtools folder

ominds, 2020-12-28:
That worked!

sn4k3, 2020-12-28:
Ok attach here the file so others with arch and manjaro can fix by just replace that file

ominds, 2020-12-28:
Sure. However, after starting the app and moments after I replied, my gnome crashed and restarted. This happened with me yesterday too while I was trying to get it working on archlinux. I think it has something to do with gnome wayland. Maybe starting it with xorg (rather than wayland) would avoid this issue. I'll check on that later. For now, here you go

[https://drive.google.com/file/d/1uXrUEdv5kCYSUkLXc9cwWyejjTyU3oF1/view?usp=sharing](url)

ominds, 2020-12-28:
Ok. Tried running it with x11, same problem. Gnome crashes moments after the application starts

Remember to change 

UVtools.runtimeconfig.json to your 5.x version, for me, that's "5.0.0-rc.2.20475.5"

And I changed UVtools.sh to run in x11 rather than wayland

`env GDK_BACKEND=x11 dotnet UVtools.dll`

However, I'm still getting the gnome crash. Here's what I have in xorg.0.log (don't know if it's relevant)

```
[     4.640] (II) Module fb: vendor="X.Org Foundation"
[     4.640] 	compiled for 1.20.10, module version = 1.0.0
[     4.640] 	ABI class: X.Org ANSI C Emulation, version 0.4
[     4.640] (EE) 
[     4.640] (EE) Backtrace:
[     4.640] (EE) 0: /usr/lib/Xorg (xorg_backtrace+0x53) [0x55eaf5877f63]
[     4.640] (EE) 1: /usr/lib/Xorg (0x55eaf5731000+0x151da5) [0x55eaf5882da5]
[     4.640] (EE) 2: /usr/lib/libc.so.6 (0x7f1b01ff2000+0x3d6a0) [0x7f1b0202f6a0]
[     4.640] (EE) 3: /usr/lib/libc.so.6 (gsignal+0x145) [0x7f1b0202f615]
[     4.640] (EE) 4: /usr/lib/libc.so.6 (abort+0x116) [0x7f1b02018862]
[     4.640] (EE) 5: /usr/lib/libc.so.6 (0x7f1b01ff2000+0x26747) [0x7f1b02018747]
[     4.640] (EE) 6: /usr/lib/libc.so.6 (0x7f1b01ff2000+0x35bf6) [0x7f1b02027bf6]
[     4.641] (EE) 7: /usr/lib/Xorg (dixRegisterPrivateKey+0x0) [0x55eaf57c6870]
[     4.641] (EE) 8: /usr/lib/xorg/modules/libglamoregl.so (glamor_init+0xc9) [0x7f1afc715fb9]
[     4.641] (EE) 9: /usr/lib/xorg/modules/drivers/modesetting_drv.so (0x7f1afc858000+0x140fd) [0x7f1afc86c0fd]
[     4.641] (EE) 10: /usr/lib/Xorg (AddGPUScreen+0x10e) [0x55eaf57aa40e]
[     4.641] (EE) 11: /usr/lib/Xorg (0x55eaf5731000+0x185eb9) [0x55eaf58b6eb9]
[     4.641] (EE) 12: /usr/lib/Xorg (0x55eaf5731000+0x1bb288) [0x55eaf58ec288]
[     4.641] (EE) 13: /usr/lib/Xorg (0x55eaf5731000+0x1bb52b) [0x55eaf58ec52b]
[     4.641] (EE) 14: /usr/lib/Xorg (InitInput+0xf5) [0x55eaf589c435]
[     4.641] (EE) 15: /usr/lib/Xorg (0x55eaf5731000+0x39798) [0x55eaf576a798]
[     4.641] (EE) 16: /usr/lib/libc.so.6 (__libc_start_main+0xf2) [0x7f1b0201a152]
[     4.641] (EE) 17: /usr/lib/Xorg (_start+0x2e) [0x55eaf576b5de]
[     4.641] (EE) 
[     4.641] (EE) 
Fatal server error:
[     4.641] (EE) Caught signal 6 (Aborted). Server aborting
[     4.641] (EE) 
[     4.641] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[     4.641] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[     4.641] (EE) 
[     4.641] (II) AIGLX: Suspending AIGLX clients for VT switch
[     5.707] (EE) Server terminated with error (1). Closing log file.
```

sn4k3, 2020-12-28:
Dont run the sh file.
Run the ./UVtools instead from linux_x64 package.
You may need to chome +x ./UVtools

sn4k3, 2020-12-29:
try with the new arch release

ampotos, 2020-12-29:
I just tried with the new release, it works for me thanks.
I use x11 with I3 on arch.

closed 2021-01-03 by sn4k3