sn4k3/UVtools #106: [FEATURE REQUEST]

RJSavell, 2020-11-29:
In the pixel dimming feature would it be possible to add a chamfer option?  for example something like this:

[image]

For example if you wanted to modify the first 5 pixels from the edge inward for first 5 layers using a pattern such as this:
layer 0: 127 127 127 127 127
layer 1: 127 127 127 127 255
layer 2: 127 127 127 255 255
layer 3: 127 127 255 255 255
layer 4: 127 255 255 255 255

Thanks,

Randy

sn4k3, 2020-11-29:
You cant use pattern to do chamfer as that would require whole image information as a pattern, by running so you would end with that pattern repeating across xy every 5 pixels.
Instead i can implement a chamfer like morph which acts over border thickness, so the input would be:

border size: 5 to 1 at 127 brightness

RJSavell, 2020-11-29:
I tried the morph / erode and it seems to work but if that is something you could do and allow a brightness setting that would be even better.

Thanks!!!!

Randy

sn4k3, 2020-12-04:
implemented

closed 2020-12-04 by sn4k3