sn4k3/UVtools #107: [BUG] Repair islands function ignores pixel setting

roofuskit, 2020-12-01:
## Describe the bug
Repair function ignores pixel settings for islands and repairs all islands regardless of size.

## To Reproduce
Steps to reproduce the behavior:
1. click issues
2. click repair
3. change "Remove islands equal or smaller to:" to 1 pixel
4. click repair.

## Expected behavior
Only islands consisting of a single pixel should be removed. 

## Screenshots
N/A

## Files
N/A

## System
**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 x64 v20H2]
  
## Exceptions Log file
N/A

## Additional context
N/A

sn4k3, 2020-12-01:
Hi, this worked for me.
My test: Draw single pixels islands, draw two pixel islands.
Use repair to remove all islands below or equal 1px. 
Result: 1px islands was removed, but 2px islands remanin untouched.

Before repair:
[image]

After repair:
[image]

Can you try this?

sn4k3, 2020-12-26:
Everything looks to work as expected

closed 2020-12-26 by sn4k3