sn4k3/UVtools #109: [BUG] Error open pws file

EriveltonMS, 2020-12-05:
## Describe the bug
Error opening .pws files

## To Reproduce
Steps to reproduce the behavior:
1. Open/ Drag a pws file to uvtools
4. See error

## Expected behavior
open the file

## System
**Desktop (please complete the following information):**
- OS: Windows 10 x64 
  
## Exceptions Log file
System.IO.FileLoadException: PREVIEW section bytes: expected 75276, got 75292, difference: -16
   at UVtools.Core.FileFormats.PhotonWorkshopFile.SectionHeader.Validate(String mark, Int32 length, Object obj) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 188
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Preview.Validate(Int32 size) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 353
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1123
   at UVtools.WPF.MainWindow.<>c__DisplayClass91_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 824

sn4k3, 2020-12-05:
Previous saved PWS files with uvtools will require discard and reslice:

From changelog:
* (Fix) PW0, PWS, PWMX, PWMO, PWMS, PWX file formats, where 4 offsets (16 bytes) were missing on preview image, leading to wrong table size. Previous converted files with UVtools wont open from now on, you need to reconvert them.

sn4k3, 2020-12-11:
Talked with Lychee team, this should be fixed on thier next release.

closed 2020-12-11 by sn4k3