sn4k3/UVtools #110: [BUG] PWMX files not opening

EnriqueGlezni, 2020-12-08:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
When trying to open a newly sliced pwmx file in version 1.4.0 I receive the error pointed below.

## To Reproduce
Steps to reproduce the behavior:
Drag file to UV tools 
Open file from menu

## Expected behavior
Open file

## Screenshots
If applicable, add screenshots to help explain your problem. Never hide information, full GUI/window screenshot is always a must!
[image]

## Files
Github 

## System
Window 10 64bit
  
Error message:
System.IO.FileLoadException: PREVIEW section bytes: expected 75276, got 75292, difference: -16
   at UVtools.Core.FileFormats.PhotonWorkshopFile.SectionHeader.Validate(String mark, Int32 length, Object obj) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 188
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Preview.Validate(Int32 size) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 353
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1123
   at UVtools.WPF.MainWindow.<>c__DisplayClass91_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 832

Please note this is a newly sliced file, has not been opened on uvtools, files were opening fine on the previous version of UV tools.

sn4k3, 2020-12-08:
Duplicate on #109 please verify if is the case

EnriqueGlezni, 2020-12-08:
Error seems to be almost the same, file type differs but I guess PWS englobes all anycubics files, but yes, seems to be the same error reported again as I saw your answer, and wanted to report that it happens to new re-sliced files, had to downgrade to 1.3.4 to be able to open any PWMX file again :D

sn4k3, 2020-12-08:
It works for me for new sliced files. Are you sure you are reslicing them and save and then open the correct file on UVtools?

[image]

EnriqueGlezni, 2020-12-08:
Yes, is a newly sliced file, so there has never been opened or manipulated on UVtools before, I keep getting the same error, I am slicing in Lychee, do not know if that can be a source for the issue, 1.3.4 working great for me now, will re-update and test again once I finish processing my urgent files, will report back if the error persists and if slicing in photon workshop works. 
Attaching file sliced on Lychee
[Ring3_Ring4.zip](https://github.com/sn4k3/UVtools/files/5661743/Ring3_Ring4.zip)

sn4k3, 2020-12-08:
Ah so that may be the cause, Lychee can also have the same bug UVtools had. As it now verify for integrity it fails to open the file. I can remove the integrity but it wont verify for errors anymore.
Open a case on lychee and forward them to this post, they are missing 16 bytes as i was

sn4k3, 2020-12-11:
Talked with Lychee team, this should be fixed on thier next release.

closed 2020-12-11 by sn4k3