sn4k3/UVtools #111: [BUG] error opening .pws file generated by Lychee Slicer

EriveltonMS, 2020-12-10:
## Describe the bug
Error opening .pws files generated by Lychee Slicer 

## To Reproduce
Steps to reproduce the behavior:
1. Open/ Drag a open a pws file generated by Lychee Slicer on  uvtools

## Expected behavior
open the file

## System
**Desktop (please complete the following information):**
- OS: Windows 10 x64 

## Exceptions Log file
System.IO.FileLoadException: PREVIEW section bytes: expected 75276, got 75292, difference: -16
   at UVtools.Core.FileFormats.PhotonWorkshopFile.SectionHeader.Validate(String mark, Int32 length, Object obj) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 188
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Preview.Validate(Int32 size) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 353
   at UVtools.Core.FileFormats.PhotonWorkshopFile.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1123
   at UVtools.WPF.MainWindow.<>c__DisplayClass91_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 832

sn4k3, 2020-12-10:
Related to #110 and #109 
Read both, lychee problem

sn4k3, 2020-12-11:
Talked with Lychee team, this should be fixed on thier next release.

closed 2020-12-11 by sn4k3