sn4k3/UVtools #113: [FEATURE REQUEST]

X3msnake, 2020-12-17:
Add mouse over information about the pixel colour 
Similar to the colour drop sampler information. ideally it could just have the value display on the information bar as one moves the mouse over the pixels alternatively it could read using a key combo or as a tool

[image]

sn4k3, 2020-12-17:
UVtools already have a brightness picker (Layer status bar with coordinates and **B** of brightness).
Hold CTRL and hover on a pixel. Release CTRL to lock the selection.
See:

[image]

X3msnake, 2020-12-17:
Awsome i think we soon need to write a manual for this :P

closed 2020-12-17 by X3msnake