sn4k3/UVtools #114: [FEATURE REQUEST] Add command line interface please

ZhenyaKa, 2020-12-21:
For example to convert grayscale png files in folder to RGB that used in Nova3d Bene4/Elfin2 Mono.

That can be used for converting files in Chitubox plugin

sn4k3, 2020-12-21:
you can use uvtools to convert formats to RGB to bene4/elfin2 mono.

ZhenyaKa, 2020-12-21:
> you can use uvtools to convert formats to RGB (bene4/elfin2 mono)
> If you just want the png files, open the .cws as zip and extract them.

I need to do it in command line mode. Chitubox stores png files in folder before packing. I need to procees that files in plugin with command line (convert mono to 3pix GRB-mode) and after that Chitubox pack it in cws.

I told about command line interface. I know, that I can do that via GUI in UVTools

sn4k3, 2020-12-21:
UVtools currently has no command line interface you can use. There was a project called UVtools.Cmd but i never developed that project.
You can easy do the conversion via a python script, you don't need a huge toolset like uvtools to do that

sn4k3, 2020-12-21:
Encode gray to rbg (repeat the 3 gray brightness bytes to a RGB):
```
var span = mat.GetPixelSpan<byte>();
var spanEncode = matEncode.GetPixelSpan<byte>();
for (int i = 0; i < span.Length; i++)
{
    spanEncode[i] = span[i];
}
```

Decode (put RBG bytes into 3 gray bytes):

```
var span = mat.GetPixelSpan<byte>();
var spanDecode = matDecode.GetPixelSpan<byte>();
for (int i = 0; i < span.Length; i++)
{
    spanDecode[i] = span[i];
}
```

As you can see is super easy to do your own script, use python.

ZhenyaKa, 2020-12-21:
> Encode gray to rbg (repeat the 3 gray brightness bytes to a RGB):
> 
> ```
> var span = mat.GetPixelSpan<byte>();
> var spanEncode = matEncode.GetPixelSpan<byte>();
> for (int i = 0; i < span.Length; i++)
> {
>     spanEncode[i] = span[i];
> }
> ```
> 
> Decode RBG to gray (put RBG bytes into 3 gray bytes):
> 
> ```
> var span = mat.GetPixelSpan<byte>();
> var spanDecode = matDecode.GetPixelSpan<byte>();
> for (int i = 0; i < span.Length; i++)
> {
>     spanDecode[i] = span[i];
> }
> ```
> 
> As you can see is super easy to do your own script, use python, 4 lines worth of code.

Thanx a lot. I realised this at Java already, but it's slower I think than your app.

sn4k3, 2020-12-21:
Is slower because maybe you are using indirect access to pixels. You need to access directly to the buffer and not safe properties which are extremly slow

ZhenyaKa, 2020-12-21:
> Encode gray to rbg (repeat the 3 gray brightness bytes to a RGB):
>
> ```
> var span = mat.GetPixelSpan<byte>();
> var spanEncode = matEncode.GetPixelSpan<byte>();
> for (int i = 0; i < span.Length; i++) [...]

And as I understand your code not some like this:

[image]

sn4k3, 2020-12-21:
My code output exactly that

ZhenyaKa, 2020-12-21:
Here my code:
Maybe Java has similar function (classes)?
public class ImageProcessor {
    public static void pix2RGB(String fileName) {
        try {
         	File file = new File(fileName);
	         BufferedImage source = ImageIO.read(file);
	         // Создаем новое пустое изображение, сжатое по короткой стороне в 3 раза
	         BufferedImage result = new BufferedImage(source.getWidth()/3, source.getHeight(), BufferedImage.TYPE_INT_RGB);
	
	         // Делаем двойной цикл, чтобы обработать каждый пиксель
		 int widthMax = source.getWidth()/3*3;
	         for (int x = 0; x < widthMax; x = x + 3) {
	              for (int y = 0; y < source.getHeight(); y++) {
	
			int pix1 = source.getRGB(x  , y)& 0xff;
			int pix2 = source.getRGB(x+1, y)& 0xff;
			int pix3 = source.getRGB(x+2, y)& 0xff;
	
	                Color newColor = new Color(pix3, pix2, pix1);
	
	                result.setRGB(x/3, y, newColor.getRGB());
	              }
	         }
	
	        // Сохраняем результат в тот же файл
	        File output = new File(fileName);
	        ImageIO.write(result, "png", output);

	        } catch (IOException e) {

	            // При открытии и сохранении файлов, может произойти неожиданный случай.
	            // И на этот случай у нас try catch
	            System.out.println("File not found or file error");
        	}

	}
    }

sn4k3, 2020-12-21:
Why you are using java? It's ugly and slow. Use python as scripting, much better.
If you want to optimize your java, see: https://stackoverflow.com/questions/39727519/efficient-access-to-image-pixels-in-java/39730660

ZhenyaKa, 2020-12-21:
matEncode.GetPixelSpan<byte>(); - I can't find any info about this functions.

sn4k3, 2020-12-21:
That is C# function of EmguCV, you can't use my functions if you are on a diferent language/framework/libraries

GetPixelSpan() returns an single array of bytes with size of width*height. So pixels are sequencial in the array, it's easy to manipulate them for this case

ZhenyaKa, 2020-12-21:
> That is C# function i did to return all pixels buffer in a array, you can't use my functions if you are on a diferent language/framework/libraries
> 
> GetPixelSpan() returns an single array of bytes with size of width*height. So pixels are sequencial in the array, it's easy to manipulate them for this case

You said about Python and I try to find this functions in it Docs :)

sn4k3, 2020-12-21:
Python:

```
img = cv2.imread('D:\Project\Capture1.jpg',0)
rows,cols = img.shape

    for i in range(rows):
      for j in range(cols):
```

OR

```
from PIL import Image
im = Image.open('um_000000.png')

pixels = list(im.getdata())
```

that last method return an array like my GetPixelSpan()

ZhenyaKa, 2020-12-21:
I just tested Java and Python. Java 2-3 times faster :)

sn4k3, 2020-12-21:
You need to find the right optimization for both languages. But if java works for you it's ok

ZhenyaKa, 2020-12-21:
> You need to find the right optimization for both languages. But if java works for you it's ok

Java posted be me earlier
Here python version (my first prog :) LOL)

`for imgFile in imgFiles:
	print (imgFile)
#	image = Image.open('slice00006.png').convert('L') 
	image = Image.open(imgFile).convert('L') 
	width, height = image.size

	pix = image.load() #Выгружаем значения пикселей.

	new_width=width//3
	print ("old width and height: ",width ,":",height)
	print ("new width and height: ",new_width ,":",height)

	pixels = list(image.getdata())

	old_pixels = list(image.getdata())
	all_pixels = []

	for y in range(height):
		for x in range(new_width):
#		r = pix[x*3  ,y]
#		g = pix[x*3+1,y]
#		b = pix[x*3+2,y]
			addr=x*3+y*width
			(b, g, r) = old_pixels[addr:addr+3]
			all_pixels.append((r,g,b))

	res = Image.new('RGB', (new_width, height))
	res.putdata(all_pixels)

	res.save(imgFile)`

sn4k3, 2020-12-21:
I'm not an active python programmer, can't help futher on optiomization, that require a bit of research (google) and testing.
Now just paralelize your work and it will be pretty fast to eat all the bitmaps

closed 2020-12-21 by sn4k3

adammhaile, 2021-01-09:
Is there a reason you dropped work on UVTools.Cmd ?
The examples discussed here are all just basic image conversion but I would really love to be able to call out to UVTools as a post-process action in PrusaSlicer so that I could automatically convert from `.sl1` to `.pwmx` format so that I don't have to do it manually in the UVTools UI.
I would be more than happy to lend dev time to make that happen. Honestly, I wouldn't even care about anything fancy just a simple format conversion would be amazing and I think a lot of people would use it so they can use PrusaSlicer.

sn4k3, 2021-01-09:
UVTools.Cmd is capable of convertion as it is.
It was a fast alternative to put UVtools working on linux and mac, but then i realize few will use it and it will be complex for some, not to mention you lost lot of functionality from the GUI.
PrusaSlicer right now dont allow you to run post-process scripts on SLA via GUI, but it does via terminal.

But at the moment you have the possibility of do your own scripts: https://github.com/sn4k3/UVtools/tree/master/UVtools.Scripts
So what you ask is easy done there

adammhaile, 2021-01-09:
@sn4k3 oh nice, the PowerShell script thing would probably work well. 
And yeah - I was just noticing there's no post-processor for SLA at the moment :( Granted, I could probably setup something that watches a folder and then runs a script against anything saved there. The PowerShell option should work nicely.
Thanks so much for the quick reply!

sn4k3, 2021-01-09:
> I could probably setup something that watches a folder and then runs a script against anything saved there
Having a service runing all the time for that i think it's not worth. You can easy do a drag and drop file to script to convert the file

adammhaile, 2021-01-09:
That's fair - I really meant only have it running while PrusaSlicer is running (something I'm already doing for other machines).