sn4k3/UVtools #115: [BUG] Crash on startup on Debian

femtostar, 2020-12-26:
## Describe the bug
Crash with null reference exception on start. All dependencies are installed and I'm trying to run the latest release. The output to the terminal is as follows:
`Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.MainWindow.UpdateMaxWindowsSize() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 466
   at UVtools.WPF.MainWindow..ctor() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 404
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 72
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0()
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.Threading.Dispatcher.EnsurePriority(DispatcherPriority currentPriority)
   at Avalonia.Threading.DispatcherTimer.InternalTick()
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 26
Aborted
`

## To Reproduce
Steps to reproduce the behavior:
1. Download latest 2.0.0 release on Debian stable (might happen on other platforms too)
2. Set UVtools binary to executable
3. Start with ./UVtools
4. Crash

## Expected behavior
UVtools starts normally
## System
**Desktop (please complete the following information):**
 - OS: Debian 10.5

sn4k3, 2020-12-26:
Can you try with the v1.4 to see if happens there too?
What GUI interface you using on debian?

femtostar, 2020-12-26:
Running Openbox on Xorg. I tried 1.4, it seems to start well enough to give me a popup window but fails with "UVtools can not run due lack of dependencies from cvextern/OpenCV". I have libopencv-dev installed, so I don't know why this is still failing.

sn4k3, 2020-12-26:
libopencv-dev won't do it. You need all those listed in: https://github.com/sn4k3/UVtools#linux

femtostar, 2020-12-26:
I did install those, except libopenexr24 which doesn't seem to be in the debian repos. I used libopenexr-dev.

sn4k3, 2020-12-26:
ok then try to find what dependency is missing, open uvtools folder in terminal, run: ldd libcvextern.so | grep not found

femtostar, 2020-12-26:
Okay, upgraded to Debian testing which resolved some dependencies (new enough glibc now for example), but it still complains about libjpeg.so.8 and libIlmImf-2_3.so.24. I have libilmbase25 and libjpeg.so.62 but libjpeg.8 and libilmbase24 aren't available as packages. Should I maybe try to build from source and hopefully configure for the lib versions available for Debian? I haven't been able to find documentation on building UVtools on Linux.

sn4k3, 2020-12-26:
Dependecies are not related to UVtools sources, so you must compile EmguCV instead.
http://www.emgu.com/wiki/index.php/Download_And_Installation

Follow the Unbutu steps, you must have build essentials and cmake installed before start the guide.
Also note that process will take about 1h depending on your machine

sn4k3, 2020-12-26:
eg:

sudo apt-get install git cmake build-essentials -y
git clone https://github.com/emgucv/emgucv emgucv 
cd emgucv
git submodule update --init --recursive
cd platforms/ubuntu/20.04
./apt_install_dependency
./cmake_configure
cmake build
cd build; make; cd ..

At this point you should able to run UVtools without dependency problems, i guess you also have to remove the confliting dependencies

femtostar, 2020-12-27:
Hello! Thanks for the tip, got that compiled and swapped in and 1.4.0 now works for me. Anything I should be trying to get 2.0 working?

sn4k3, 2020-12-27:
I guess that may due your desktop environment, you still get same error with v2?

femtostar, 2020-12-27:
Yes, same NullReferenceException, as you say seemingly related to window management. I know you use/develop/test on Windows but have any other users gotten 2.0.0 working on Linux+X11 yet?

sn4k3, 2020-12-27:
i tested on Unbutu and Mint, both work

sn4k3, 2020-12-27:
Try this custom build: https://we.tl/t-30e1B37x9o

femtostar, 2020-12-27:
Well, that gets a little further. Now it's able to give me the cvextern error popup (I wonder if the release build's Emgu build is actually working for me and that's why I don't see it there?), swapping in the same fixed Emgu build that works for me on 1.4.0 seemingly fixes that but it briefly draws a blank window before crashing with a similar error to the release build.

```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.MainWindow.UpdateMaxWindowsSize() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 457
   at UVtools.WPF.MainWindow.OnOpened(EventArgs e) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 465
   at Avalonia.Controls.Window.ShowCore(Window parent)
   at Avalonia.Controls.Window.Show()
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 26
Aborted
```

sn4k3, 2020-12-27:
And on v1.4 are you able to open the setting window (F12?)

sn4k3, 2020-12-27:
Another build: https://we.tl/t-G5tS7BdMMh
Run that from terminal and send me the output messages

femtostar, 2020-12-27:
Can confirm, opening settings crashes 1.4 with:
```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.Windows.SettingsWindow..ctor() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Windows\SettingsWindow.axaml.cs:line 64
   at UVtools.WPF.MainWindow.MenuFileSettingsClicked() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 682
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__139_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass8_0.<Post>b__0()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken)
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 28
Aborted
```
New custom build crashes (after fixing Emgu as with before) with:
```
Settings is null?: False
Settings.General is null?: False
Settings.General is null?: False
Screens is null?: False
ScreenCount: 2
Screens.Primary is null?: True
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.MainWindow.UpdateMaxWindowsSize() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 463
   at UVtools.WPF.MainWindow.OnOpened(EventArgs e) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 471
   at Avalonia.Controls.Window.ShowCore(Window parent)
   at Avalonia.Controls.Window.Show()
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 26
Aborted
```

sn4k3, 2020-12-27:
Ok, found the problem, Screens.Primary doesnt exists for you. You have two screens but none is set as primary

femtostar, 2020-12-27:
Yes, I do have two displays. TIL LXDE's monitor configuration tool doesn't set one as primary. I might be able to work around this on a single-display machine for now, but would it be possible to default to arbitrarily setting the first screen as, for the purposes of UVtools, "de-facto primary" if the DE does not provide a primary display?

sn4k3, 2020-12-27:
This should fix: https://we.tl/t-nwO0MhaRnN
Use UVtools on your first monitor if resolution from the 2nd is diferent in any way

femtostar, 2020-12-27:
Both same res, but will keep that in mind. That works, thanks! Should I close this or should I wait for a patch in a release?

sn4k3, 2020-12-27:
Leave it open, i will close on patch release

sn4k3, 2020-12-27:
Can you execute this commands on terminal and paste back the result?

xrandr
xrandr --version
xrandr --listproviders
xrandr --listmonitors

femtostar, 2020-12-27:
Yep, here you go

> xrandr
```
Screen 0: minimum 320 x 200, current 7680 x 2160, maximum 16384 x 16384
VGA-1 disconnected primary (normal left inverted right x axis y axis)
DP-1 connected 3840x2160+3840+0 (normal left inverted right x axis y axis) 621mm x 341mm
   3840x2160     60.00*+  30.00    30.00    24.00    29.97    23.98  
   1920x1080     60.00    50.00    59.94    30.00    24.00    29.97    23.98  
   1920x1080i    60.00    50.00    59.94  
   1680x1050     59.95  
   1280x1024     75.02    60.02  
   1440x900      59.89  
   1280x960      60.00  
   1280x800      59.81  
   1152x864      75.00  
   1280x768      59.87  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       75.00    72.81    66.67    60.00    59.94  
   720x400       70.08  
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DP-2-1 connected 3840x2160+0+0 (normal left inverted right x axis y axis) 621mm x 341mm
   3840x2160     60.00*+  30.00    30.00    24.00    29.97    23.98  
   1920x1080     60.00    50.00    59.94    30.00    24.00    29.97    23.98  
   1680x1050     59.95  
   1280x1024     75.02    60.02  
   1440x900      59.89  
   1280x960      60.00  
   1280x800      59.81  
   1152x864      75.00  
   1280x768      59.87  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       75.00    72.81    66.67    60.00    59.94  
   720x400       70.08  
DP-2-2 disconnected (normal left inverted right x axis y axis)
DP-2-3 disconnected (normal left inverted right x axis y axis)
```
> xrandr --version
```
xrandr program version       1.5.0
Server reports RandR version 1.5
```
> xrandr --listproviders
```
Providers: number : 1
Provider 0: id: 0x4a cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 0 name:modesetting
```
> xrandr --listmonitors
```
Monitors: 2
 0: +DP-1 3840/621x2160/341+3840+0  DP-1
 1: +DP-2-1 3840/621x2160/341+0+0  DP-2-1
```

sn4k3, 2020-12-27:
Read here: https://github.com/AvaloniaUI/Avalonia/issues/5228#issuecomment-751508296

femtostar, 2020-12-27:
Interesting. My display setup is somewhat unusual - a laptop with no internal display on a docking station running one display from a port on the laptop itself and one display over the docking station. Further, it seems to have decided the VGA port, which isn't in use, is the primary display. Avalonia/UVtools should probably handle this more cleanly but I do recognize my setup is a bit of a corner case in terms of display configuration.

sn4k3, 2020-12-27:
I guess it will depend on OS and combos.
Right now it's fixed by just using first screen on this list

sn4k3, 2020-12-28:
You can set DP-1 as primary by running `xrandr --output DP-1 --primary`

closed 2020-12-29 by sn4k3