sn4k3/UVtools #116: [FEATURE REQUEST] MacOS .app bundle

jorgerobles, 2020-12-27:
## Is your feature request related to a problem? Please describe
Nope

## Describe the solution you'd like
Want to have the application as a single bundle on MacOS, like the other apps

## Describe alternatives you've considered
No.

## Additional context
Maybe this helps: https://www.xamboy.com/2020/05/28/net-core-application-macos-packaging-and-notarization/

sn4k3, 2020-12-27:
See this: https://github.com/dirkessl/uvtoolsmacappcreator

closed 2020-12-28 by jorgerobles

laird, 2021-01-03:
The generated app bundle isn't notarized, making it complex to run, increasingly so for newer versions of MacOS. This is becoming standard for Mac Apps, as a means of reducing the spread of unsafe apps. There's no cost, just the complexity of setting it up as a part of the build process.

sn4k3, 2021-01-03:
"Notarization is required for apps distributed outside the app store starting in macOS 10.15 (Catalina), and you'll have to enable hardened runtime and run codesign on your .app in order to notarize it successfully.
**You'll need a Mac computer for this step, unfortunately, as we have to run the codesign command line tool that comes with Xcode.**"
"In order to sign your files, you need an Apple developer account. In order to notarize your app, you'll need to do the following steps with a Developer ID certificate, which requires a paid Apple developer subscription."

Can't do it in a automated build process, i don't own a mac and VM is buggy, laggy and painfull to work with.
Also requires a valid Developer ID certificate which is paied.
So this won't be worked out.
All i can do is upload the zip package as a folder like before. Or just trust the app on mac security