sn4k3/UVtools #117: Can I do an inner erosion?

ominds, 2020-12-28:
I would like to do an erosion on the inside rather than the surround of my layers. To give you an idea about the problem I'm facing, I have a cylinder with inner diameter 30mm, and wall thickness is 2mm, making the outer diameter 34mm. However, when I print (exposure is optimum), I get 34mm outer diameter (correct), and 2.36mm wall thickness (this is +0.36mm). If I'm able to erode the inner pixels by 5-6 pixels (I have a mono x), then this should solve the problem. Is this possible with UVtools? I'm attaching some pictures to further explain the problem

Outer diameter looks good
[image]

Wall thickness very off
[image]

Thanks in advance!

sn4k3, 2020-12-28:
No. Erode will not only erode external but all contours, on your case the outter and inner hole will get eroded. You cant excude externals.
I guess your problem is with resin expansion, you can dim the model in a attempt to address that.

1. Open pixel dimming
3. set brightness to eg. 200
4. Hit "Solid"
5. Apply
6. Try print and measure

ominds, 2020-12-29:
Can you kindly explain to me what this will do? I understand with monochrome screens (photon x), it's black/white pixels. White gets exposed to UV, black doesn't. How does dimming and changing it to gradient affect this?

sn4k3, 2020-12-29:
It's not only black/white pixels. you have a range from 0 to 255 on color and mono lcds, where 0 is black and 255 is white, all in between are gray shades, eg: 127/128 = half of white. So it's not only 0% or 100% but all in between are usable, you have 255 tones to use. Dimming will undercure that pixels and on big masses it will avoid expansion. But you need to find your right value, in other words you need to print and test with diferent values up to find the right one for your combo

ominds, 2020-12-29:
Ok. However, and this is really puzzling me. My outside diameter is correct. If there's expansion happening, it's only happening inwards by 0.36mm. I don't know what could cause such an inward-only expansion.

ominds, 2020-12-29:
If expansion happened in every direction, I would expect the Outer Diameter to be larger than 34mm

sn4k3, 2020-12-29:
These things aren't linear, physics are involved, and it will do diferent for every printer/lcd/led/resin/temperature/others. The outside diameter is wider and less constrained, centers have more tension and are often more affected by this problem

ominds, 2020-12-29:
But I'd say that trimming all cavities from inside would solve the problem. 6*0.05 = 0.3mm

ominds, 2020-12-29:
Because I have pipes that slide into each other. Very precise parts. The way this is going, I can't use this machine for precise parts. Currently, the pipes simply aren't fitting inside each other

sn4k3, 2020-12-29:
If you suspect about slicer puting the wrong dimentions, use UVtools to generate a Tolerance test, there you can set the outer and inner diameter and it will tell you the expected size. This can confirm if slicer is doing right or not. 

If you want to trim the inside you can use pixel editor with a large circle brush and remove a portion of the center, still will be hard to do it perfectly at center.

About these file formats nothing are a standard yet and they keep changing, you can use UVtools library to handle without have to rewrite, otherwise you can look at uvtools sources.

sn4k3, 2020-12-29:
> Because I have pipes that slide into each other. Very precise parts. The way this is going, I can't use this machine for precise parts. Currently, the pipes simply aren't fitting inside each other

Every machine can be used for precise printing, you just need to tune or find the offset for your problem. For example on FDM i always give 0.5 to 0.6mm tolerance for parts that must fit and that does the trick. Due various factors (said above) you can't expect a design to print exacly at same dimentions, they often shrink or expand, and you need to design taking that in account

closed 2021-01-03 by sn4k3