sn4k3/UVtools #121: [BUG] Installation path on update changed

taker218, 2021-01-04:
## Describe the bug
I installed the UVtools in another directory, rather then the standard one (program files). When I run an update it always populates the path with the standard one, not the one I installed the program before.

## To Reproduce
Steps to reproduce the behavior:
1. Install older version of UVtools to a non standard folder
2. Perfom update from the UVtools UI

## Expected behavior
I would have expected to have my current installation path in the update installer, not the standard one.

## Screenshots
[image]
[image]

## Files
Were possible attach the problematic file, both source and output file if applicable.

## System
**Desktop (please complete the following information):**
 - OS: Windows 10 x64 20H2
  
## Exceptions Log file
In some unexpected exceptions errors will generate and save a log file under appfolder/Logs/AppErrors.
If present you can attach the log file here. Check file name against date/time to be sure it's related to your problem.

## Additional context
Add any other context about the problem here.

sn4k3, 2021-01-07:
Each instalation is like a fresh install. The path is not saved on register and then no way to pull back the directory.
The msi wix is really a pain to work with...

sn4k3, 2021-01-07:
Fixed, you need to remove uvtools, install with the new version and from now on it will save the path

closed 2021-01-07 by sn4k3