sn4k3/UVtools #122: [BUG] Opening some .photon files from the File->Open menu crashes the app, drag and drop works

csakip, 2021-01-10:
Trying to load some .photon files from File menu -> Open crashes the app every time. Opening the same file via drag and dropping opens it fine.
Not all .photon files do this. Some are sliced in chitubox, some are in z-suite and there's no correlation to the slicer.

### File
[2Rim.zip](https://github.com/sn4k3/UVtools/files/5792479/2Rim.zip)

## System
Win 10

## Error log 
<hr>

There are two errors in event log.
### Application error:

Faulting application name: UVtools.exe, version: 2.2.0.0, time stamp: 0x5fbf100e
Faulting module name: KERNELBASE.dll, version: 10.0.18362.1237, time stamp: 0x71e81044
Exception code: 0xe0434352
Fault offset: 0x0000000000043b29
Faulting process ID: 0x2248
Faulting application start time: 0x01d6e73c0db45a9c
Faulting application path: C:\Program Files (x86)\UVtools\UVtools.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report ID: 5c9bd6e0-1a48-497a-97cb-5338efee880e
Faulting package full name: 
Faulting package-relative application ID: 

<hr>

### .NET Runtime

Application: UVtools.exe
CoreCLR Version: 5.0.120.57516
.NET Version: 5.0.1
Description: The process was terminated due to an unhandled exception.
Exception Info: System.InvalidCastException: Specified cast is not valid.
   at Avalonia.Win32.Interop.UnmanagedMethods.IShellItemArray.GetItemAt(UInt32 dwIndex, IShellItem& ppsi)
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass1_0.<ShowFileDialogAsync>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at UVtools.WPF.MainWindow.OpenFile(Boolean newWindow) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 710
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 26

sn4k3, 2021-01-10:
Hi, open your file via File - Open works fine for me, did you send the right file?

csakip, 2021-01-14:
It works now. I've updated to the latest UVTools.

closed 2021-01-14 by csakip

sn4k3, 2021-01-14:
Nothing changed for photon file. If you are loading from pen drive that can be the problem