sn4k3/UVtools #124: Maximum island fix based on size

eried, 2021-01-13:
I remember that the limit of 255 in "Remove islands equal or smaller than:" was because it might be destructive, but in big sculptures it does not make any sense, i.e.:

[image]

This Knight it is 13 cm height and since it was sculpted manually (painfully slow) there is a lot of islands of 300-400 pixels that can be removed because they are just glitchy 2-3 layer triangles, so, again that arbitrary limit of 255 does not make any sense.

A limit based on the size of the object OR just leaving the user to decide higher values makes a lot more sense, the user already knows all the fix tool might be destructive.

sn4k3, 2021-01-13:
255 is already huge, it's about 1.3cm, if you use it as no limit it can really make bad things on your model. As you said glitch masses can be removed but it will also remove everything else, with no iteration limit it can do a hole from bottom to top.
But if you find a good use for that values i can increase the limit to about 35k

eried, 2021-01-13:
255 is an extremely tiny area:

[image]

anything sculpted may have some dangling features like that

sn4k3, 2021-01-13:
Ok, i will increase the limit, is 65535 ok?

closed 2021-01-13 by sn4k3