sn4k3/UVtools #125: [FEATURE REQUEST] Lift speed and distance

bttguy, 2021-01-16:
## Is your feature request related to a problem? Please describe
Cant find the settings for lift speed and lift distances.

## Describe the solution you'd like
I would like to change them within UVtools if possible.

## Describe alternatives you've considered
I also prefer to use PrusaSlicer for my 3rd party SLA printer, so your tool is very helpful, and I mean it, when I say its kinda like a pro grade official product :) 
But since PS is just ment for their own SLA printer, which works with a tilt mechanism instead of lift, I cant find any way to set lift speed/distance neither within the slicer and neither within UVtools. Am I missing something, or there is currently no way of doing that. Or this is irrelevant to do so, and the predefined values are just good. I see those within UV tools, just cant find a way to modify them.

sn4k3, 2021-01-16:
What is your printer model?

bttguy, 2021-01-16:
Creality LD-002H

sn4k3, 2021-01-16:
Just import printer from UVtools into prusa slicer, look under printer -> Notes.
On UVtools you can change parameters on Tools -> Change print parameters

[image]

bttguy, 2021-01-16:
Ohh, so thats the place for it! Thank you! Keep up the great work! :) 
Should I close this FR than?

sn4k3, 2021-01-16:
Yep, that information was all on the readme (https://github.com/sn4k3/UVtools#install-and-configure-profiles-under-prusaslicer) and profile importer tool.
[image]

closed 2021-01-16 by sn4k3