sn4k3/UVtools #126: [BUG] uvtools crashes when trying to do a xyz calibration

mylife4aiurr, 2021-01-16:
## Describe the bug
Like the title. Uvtools crashes when attempting to use this save as base for uvtools calibration file.

## Files
https://drive.google.com/file/d/1u04tMyhFTOKkKSww2xVR755jhz9w1lpx/view?usp=sharing

## System
 - OS Windows 10
  
## Additional context
I have not used prusa slicer for a while. Save file was from an old version of Prusa slicer. All my clear resin profiles were done on prusa sl1. I looked for a prusa save file so uvtools could build a calibration model. 
Until fixed I'll make a new dummy save in new version of prusaslicer and report if crash still occurs.

mylife4aiurr, 2021-01-16:
Doesn't happen with new prusaslicer v2.3.0 saves

sn4k3, 2021-01-16:
Malformed file, prusaslicer.ini is missing

[image]

mylife4aiurr, 2021-01-17:
Yea - old Prusa version save format. Came across it and letting you know , uvtools didnt like it. 

Long as user use current version of prusa slicer they'll never see this problem. Gonna close this

closed 2021-01-17 by mylife4aiurr