sn4k3/UVtools #128: [BUG]

pesalek, 2021-01-21:
## Describe the bug
XYZ accuracy test does not produce hollow model with walls

## To Reproduce
Create XYZ accuracy test, hollow model tick box reacts in preview, but after generating, it is always a square with or without release hole. Not a square box with walls.

[image]

[image]

sn4k3, 2021-01-21:
It works ok for me, preview shows you the last layer only, after generate you are showing the first layer which is solid due no relif hole, if you go up on layers you will get the walls
[image]

sn4k3, 2021-02-01:
closing, due lack of anwser

closed 2021-02-01 by sn4k3