sn4k3/UVtools #130: [BUG] Lights-off second longer then what's set in Lychee/Chitubox

ctaatgen, 2021-01-22:
## Describe the bug
Light-off Delay for bottom/normal layers is set to 7 seconds in Lychee. However when looking at Print Parameters in UVTools it is much longer (14.40 and 12seconds)

## To Reproduce
Steps to reproduce the behavior:
1. Set Light-off Delay to 7 seconds in Lychee resin settings.
2. Export to CTB (in my case for Phrozen Sonic mini 4k
3. Open in UVTools and check Print Parameters

## Expected behavior
I would expect the times to be the same

## Screenshots
https://i.imgur.com/rHldJDX.jpg
https://i.imgur.com/OlxQKMW.jpg
https://i.imgur.com/RDgJ4gx.jpg

## System
Windows 10 x64

ctaatgen, 2021-01-22:
I also tried Chitubox, which has the same kind of discrepancy (9/9 seconds becomes 14.40 and 12 seconds)

sn4k3, 2021-01-22:
That's a new feature of v2.3.1, UVtools auto set that values.
Go to Settings (F12) -> Automations and set both light off to 0.
Follow release changelogs to know whats new or why things changes

closed 2021-01-22 by sn4k3