sn4k3/UVtools #131: [FEATURE REQUEST] Variable Slice calculator

X3msnake, 2021-01-23:
Can you port and implement in UVTools a tool similar to the old Ember (autodesk) Varislice Tool?

I posted about this here
https://github.com/prusa3d/PrusaSlicer/issues/3191

sn4k3, 2021-01-23:
Not impossible but i dont see a viable way to do it on UVtools without loss, this must come from slicer with 3D model, with 2D images original information is gone and all chamfers are choped by that same height increments. In addition which printers can support that? Maybe true gcode printers? And chitubox using position z per layer?
PrusaSlicer already have that for FDM, porting to SLA would be "easy" but still requires some work, i guess it would require building a table where user must set fixed usable heights and corresponding exposure times for each. The major problem is i don't see they support that for current SL1 anytime soon, as that also require a firmware change and a change of file format with a manifest file. Currently all printers i know are very limited on this area, layer height always fixed and thier firmware are made to calculate things arround that value.
Well if most of file formats can support that, lychee as generic slicer for sla would be a better bet with faster implemention

X3msnake, 2021-01-23:
Err... ctb v3 supports this. it has always supported z position and it now supports per layer exposure.

the way to make this work on UVTools is the same it works for varislice.

you slice a file at the lowest layer size you want to use, feed the script the STL and discard not used layers and ajust exposure per layer based on a pre-determined best exposure table

any slicer can do the first part

sn4k3, 2021-02-06:
Implemented

closed 2021-02-06 by sn4k3