sn4k3/UVtools #135: [BUG] PhotonS save error

lcoronelp, 2021-01-27:
## Describe the bug
Saved photons file doesn't recognized by PhotonS machine (neither open in Chitubox or UVTools)

## To Reproduce
Steps to reproduce the behavior:
1. Open Photons file exported from Chitubox
2. Execute Morph Erode 3/1 Chamfer: False Layers: Bottom
3. Save as PhotonS File
4.1. Try to print in PhotonS Machine -> Error: not recognized file
4.2. Try to open in Chitubox: show as [Fig. 1](#user-content-fig1) Only one triangle and don't calculate correct the time
4.3. Try to open in UVTools again: show as [Fig. 2](#user-content-fig2) Error in operation (overflow)

## Expected behavior
Print without problem

## Screenshots
<a id="fig1"></a>
[image]
[Fig. 1]

<a id="fig2"></a>
[image]
[Fig. 2]

## Files
[Gatitos.zip](https://github.com/sn4k3/UVtools/files/5881468/Gatitos.zip)
Chitubox Project

[Gatitos.zip](https://github.com/sn4k3/UVtools/files/5881466/Gatitos.zip)
File Exported from Chitubox

[GatitosEroded.zip](https://github.com/sn4k3/UVtools/files/5881473/GatitosEroded.zip)
File saved from UVTools after eroded

## System
 - OS: Linux Xubuntu 20.04 XFCE
  
## Errors when opened in UVTools again
```
System.OverflowException: Arithmetic operation resulted in an overflow.
   at UVtools.Core.FileFormats.PhotonSFile.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonSFile.cs:line 502
   at UVtools.WPF.MainWindow.<>c__DisplayClass95_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 921
```

Thanks Tiago! If you need more info or test something, não ha problema, é só dizer :)

sn4k3, 2021-01-27:
PhotonS is not write-supported by UVtools. Is a read-only format.
You should use pws instead

lcoronelp, 2021-01-27:
Thanks for the fast response!

Then i have two problems:
- When i open the PhotonS file in UVTools, i can't save to pws (only photons)
- Chitubox doesn't export in pws
- The convert function in UVTools doesn't work (i think is not implemented yet not?)

Know some workflow to use Chitubox and UVtools to make PWS files?

Thanks again!

sn4k3, 2021-01-27:
Open the photons on UVtools, go to File -> Convert -> PhotonS (PWS)

lcoronelp, 2021-01-27:
Convert as you say doesn't do anything (nor shows error)

Have some log that can i check?

sn4k3, 2021-01-27:
Can you send me your photons file? use wetransfer and put the link here

lcoronelp, 2021-01-27:
Video of trying to convert:
https://user-images.githubusercontent.com/1972410/106024584-d4ca9b80-60bf-11eb-977c-7e9d50d70aff.mp4

The .photons is in the bug report in a zip. https://github.com/sn4k3/UVtools/files/5881466/Gatitos.zip

The version of uvtools is 2.3.2 X64

Thanks!

sn4k3, 2021-01-27:
Works for me, this is the output:
[Gatitos (2).zip](https://github.com/sn4k3/UVtools/files/5881758/Gatitos.2.zip)

If you try to convert to ctb, does it works?

lcoronelp, 2021-01-27:
Didn't work, the convert menu doesn't work at all

And i note now than if i move the mouse in the menu, it closes immediately.
https://user-images.githubusercontent.com/1972410/106026229-79011200-60c1-11eb-9fd1-85cd75c772d3.mp4

Some bad interaction with the below preview? I note also that is not showing correctly

I have Nvidia GeForce GTX 1080 with OpenGL ES 3.2 NVIDIA 450.102.04

Curious.

sn4k3, 2021-01-27:
What OS?

sn4k3, 2021-01-27:
Try this build: https://we.tl/t-5xPqoU3Gp0

lcoronelp, 2021-01-27:
Ok this version works as expected (is this a Qt version?)

The preview keeps rendering badly (but is a minor bug)
And sometimes it closes unexpectedly (two times making different actions)

Thanks for the help!

sn4k3, 2021-01-27:
Not Qt, this is AvaloniaUI. A recent framework, and yes, it have some bugs, but as they release new versions is getting better. Windows have the least bugs.
I also notice that light desktops are worst with it, gnome seens more solid with this ui framework

lcoronelp, 2021-01-27:
Great! i didn't know it!
If you need test or anything that i can do, tell me!

Thanks again! Your work in this app is amazing!

sn4k3, 2021-01-27:
Can you send a screenshot of how main window and other window (for example Tool - Morph) look on your os? The patch i send you i have redesigned the UI.

lcoronelp, 2021-01-27:
[image]
[image]
[image]

lcoronelp, 2021-01-27:
[image]
Repair error process

sn4k3, 2021-01-27:
Looks sharp, only that message box still have hidden text :(

lcoronelp, 2021-01-27:
The size of the message box is automatic? or calculated by the inner text?

If i resize it, the next times, it keeps extended
[image]

sn4k3, 2021-01-27:
Automatic and limited to screen size, but sometimes it line breaks and clip words. It's a bug on the control and also happen with windows os

lcoronelp, 2021-01-27:
Ouch. bugs bugs bugs everywhere xD

Im web developer, in Avalonia don't have something similar to white-space:nowrap? I thing that this message is short (or break in two lines always?) Sorry if you was tried this yet :P

sn4k3, 2021-01-27:
It's really a problem on controler. It should wrap automaticly if needed, but all i see it dont need to wrap still it line break without a motive, worse, it clips the text. I already open a issue on controler but no response so far. But yes this framework have lot of bugs to fix, if you go to thier github the issues are flood. But they are in good path

lcoronelp, 2021-01-27:
Sorry for the late answer. I was working (you know).

I see that Avaloina have only few months! But with a great and fast progress!

I like ask you, I can pass some option to the executable UVTools or something to obtain debug info to send to you? (because it's closing unexpectedly in random points)

sn4k3, 2021-01-27:
It dont generate debug info. All you can do is execute from terminal and when it crash see if it return any message to the terminal

lcoronelp, 2021-01-27:
Only
```
1193721 segmentation fault  ./UVtools
```

Checking a top i can see multiple process in hierarchy.
[image]

Inclusive after the segmentation fault without any UVTools windows open :O
[image]

sn4k3, 2021-01-27:
you can try to compile the libcvextern on your system and replace by yours, it can be...

lcoronelp, 2021-01-27:
Compiling.

By the way, unfortunately neither the pws exported by the latest version nor the one you sent me work on Anycubic PhotonS :(

The machine show "File format error" when i touch play button.

sn4k3, 2021-01-27:
You need to update your printer firmware

lcoronelp, 2021-01-27:
I have the newest firmware 5.0.2, i can print correctly any pws exported with Photon workshop

sn4k3, 2021-01-27:
Can you try slice by PrusaSlicer with my printer installed? Maybe the convertion from PhotonS lack of important parameters

lcoronelp, 2021-01-27:
I will try (i currently don't have PrusaSlicer installed)

BTW, compiled and copied the libcvextern.so.

The preview keeps rendering badly
The app keep closing unexpectedly

:(

I send to you my libcvextern.so if it works for other: Xubuntu 20.04 https://we.tl/t-r6FnwzLHBn

lcoronelp, 2021-01-30:
Hi again, i installed PrusaSlicer but i had some problems:

- UVtools only find the Prusa Slicer folder at `%HOME%/.PrusaSlicer` instead `%HOME%/.config/PrusaSlicer` defined in Prusa Slicer 2.3.0

- I make an `ln -s ~/.config/PrusaSlicer ~/.PrusaSlicer` and the UVTols recognize the folder correctly

- In my version (Avalonia) the buttons "unselect all" and "select all" in profiles didn't work

- Exporting from PrusaSlicer and importing in UVTools, the pws file works perfectly and is recognized by the printer without problem.
I think that the problems is in the conversion from Chitu .photons to .pws

- I noticed that the preview of file from PrusaSlicer work corretly, the Chitubox file don't.

I hope this can be useful!

lcoronelp, 2021-01-30:
Hi again, I can see that my version of UVTools is very inestable:

With the same file (exported with same config from PrusaSlider), sometimes imports correctly and sometimes closes unexpectedly.

Can be because the avalonia version?

lcoronelp, 2021-01-30:
I testing ten times the same file in non Avalonian (gtk?) version (official version) and don't close any moment. ;)

lcoronelp, 2021-01-30:
Some interesting detail if that the avalonian version, the texts are bigger than the official version
[image]
Left: Avalonian
Right: Official

lcoronelp, 2021-01-30:
Another thing:

In PrusaSlicer Printer Notes if i don't add `LiftSpeed_120` and only keep the `LiftingSpeed_120`, then the UVTools don't recognize the lifting speed and say (i think the default) 100mm/min

Additional, i can add `BottomLiftSpeed_65` to add the bottom lift speed (as Chitubox), and this config show correctly on UVTools for bottom layers.

I don't see any list of custom properties in the readme, do you have some info about that?

Thanks!

sn4k3, 2021-01-30:
> Hi again, i installed PrusaSlicer but i had some problems:
>
> * UVtools only find the Prusa Slicer folder at `%HOME%/.PrusaSlicer` instead `%HOME%/.config/PrusaSlicer` defined in Prusa Slicer 2.3.0
> * [...]

Thanks for the report i will work on that. 
In any case you should not use chitubox with photons since it have no AA and incomplete file info.
You can use chitubox if you create a custom printer and use Zip or ctb format and then convert with UVtools.

> With the same file (exported with same config from PrusaSlider), sometimes imports correctly and sometimes closes unexpectedly.
> Can be because the avalonia version?

Both are avalonia, the diference is one is the RC1 version and the last i sent you is the v1 "stable"
As told this UI have some bugs and can be unstable under some desktops, if you can try gnome i guess your problems will gone.

> Some interesting detail is that in the avalonian version, the texts are bigger than the official version

I retouched the GUI, thats why.

> In PrusaSlicer Printer Notes if i don't add LiftSpeed_120 and only keep the LiftingSpeed_120, then the UVTools don't recognize the lifting speed and say (i think the default) 100mm/min

Oh, i miss that one, it should be LiftSpeed_xxx will fix it next version.
Printer notes already show all compatible variables with the format, but you can find them here:
https://github.com/sn4k3/UVtools#custom-printer-notes-keywords-for-prusaslicer

lcoronelp, 2021-01-30:
Thanks for the info Tiago!

If you need test something in no Gnome environment (XFCE with nvidia card) and/or PhotonS machine or some other test that you may need and you think i can do it, only write me.

Keep in touch!

lcoronelp, 2021-01-30:
I see another small thing (do you prefer that i submit another bug when i found some detail? or we talk in another space? or i can mix all here?)

When import SL1 into UVTools, if the file have an dot `.`, then the PWS exported is trimmed at this point (probably the slice function is slicing for the first dot instead the last or the regex is not working as expected :D)

sn4k3, 2021-01-30:
That problem is already fixed on current dev. Thanks for tell me back

yaroslaw, 2021-02-03:
I have exactly same issues with printer not recognizing PWS after UVTools (win64). If it's fixed, where can I get the latest binaries?

sn4k3, 2021-02-03:
> I have exactly same issues with printer not recognizing PWS after UVTools (win64). If it's fixed, where can I get the latest binaries?

Also converted from photons?

yaroslaw, 2021-02-04:
Yeap. Sliced in Chitubox (both 1.8 beta and then 1.7 to confirm). First
tried to save a photons (there's no clue it shouldn't work in an app, and
it tries to save things), then saw a .pws export so used it also. Preview
picture is corrupted every time, and printer (5.0.2 updated) says it
doesn't recognize file format.

Also (tried only one model) in trying to fix resin traps it made perfectly
good hollow model with a drain hole completely "unhollow". Do you need a
model.photons for troubleshooting?

sn4k3, 2021-02-04:
As said on this topic, uvtools don't fully support PhotonS file format, is for read-only, any attempt to write or convert will fail. You need to slice to PWS on slicer, If you fully read this topic from begin you will find the same problem and "resolution" which is: dont use photons as source

yaroslaw, 2021-02-04:
Chitubox don't have an option to export to any other format then .photons
for PhotonS. Is there a workaround you suggest?

Thanks!

sn4k3, 2021-02-04:
If you want to use chitubox you need to create a custom printer and coping same properties from photons. Then you can export to zip and convert with uvtools to pws.
Or use prusaslicer or lychee

yaroslaw, 2021-02-05:
Thanks, I'll check it out.

sn4k3, 2021-02-06:
PhotonS still readonly, a error will pop if anyone try to write a photons file.

closed 2021-02-06 by sn4k3