sn4k3/UVtools #137: [QUESTION] gcode executed by ctb file

cubiq, 2021-02-05:
Is it possible to know what gcode is actually executed when you run a ctb file? I tried to export into a chitubox zip that contains a run.gcode file, I tried to run it but it doesn't seem to do everything that is needed to actually print. Specifically the Z axis raises and lowers but the screen doesn't seem to be updated and the UV light doesn't turn on.

Thanks!

sn4k3, 2021-02-05:
Ctb don't use gcode, it follows a data table inside the file. You can modify that table with UVtools but per layer data is only changeable if your printer supports that feature, most don't

cubiq, 2021-02-05:
thanks for your reply. Is the initial data table sent as gcode or is it interpreted by the machine firmware at print start?

sn4k3, 2021-02-05:
Again no gcode is sent. All data is interpreted from the table, you can see the table on UVtools Properties table. And per layer table at UVtools Layer Data table.
ctb printers takes gcode files only to update EEPROM values and some hacks, nothing more (no printing with them)

sn4k3, 2021-02-05:
If you want to know more about the format you have this docs: https://github.com/cbiffle/catibo/blob/master/doc/cbddlp-ctb.adoc

cubiq, 2021-02-05:
thanks a lot for your reply, it's all very frustrating :) I kinda hate these machines :D

sn4k3, 2021-02-05:
Sadly most fileformats are capable of change many parameters even per layer but most printers firmware dont use them and use only the globals values. Only a few printers can benefit from the actual format:

Phrozen Sonic Mini 4K
Phrozen Sonic 4K
Elegoo Saturn
Elegoo Mars 2 Pro
Nova3D Bene 4
Nova3D Elfin2 Mono SE

closed 2021-02-05 by cubiq