sn4k3/UVtools #141: [FEATURE REQUEST] Allow MaterialNotes to override settings from PrinterNotes

nvx, 2021-02-11:
## Is your feature request related to a problem? Please describe
The automation to automatically convert SL1 files to the desired output using the PrinterNotes data is amazing, but some settings may be material-specific that are not natively supported in the SL1 file. For example lift heights, lift speeds and light off delay, some more viscous resins require higher lift heights, slower lift speeds, and longer light off delays to allow the resin to flow.

## Describe the solution you'd like
Any settings in MaterialNotes field in the SL1 file should be used instead of the PrinterNotes settings.

For example with this PrinterNotes:
```
Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_PRUSA3D
PRINTER_MODEL_SL1
PRINTER_VENDOR_ANYCUBIC
PRINTER_MODEL_PHOTON
FILEFORMAT_PHOTON

START_CUSTOM_VALUES
LightOffDelay_10
BottomLightOffDelay_20
BottomLiftHeight_10
LiftHeight_5
BottomLiftSpeed_40
LiftSpeed_65
RetractSpeed_150
BottomLightPWM_255
LightPWM_255
END_CUSTOM_VALUES
```

And this MaterialNotes:
```
START_CUSTOM_VALUES
BottomLiftHeight_10
LiftHeight_10
BottomLiftSpeed_40
LiftSpeed_40
END_CUSTOM_VALUES
```

I'd expect the converted photon file to have a LiftHeight of 10mm, and a LiftSpeed of 40mm/m for both regular and bottom layers. While other materials with an empty MaterialNotes would simply use the PrinterNotes default.

## Describe alternatives you've considered
Manually editing the parameters, but this is cumbersome and prone to human mistakes.

sn4k3, 2021-02-11:
Yeah thats good idea, will work like that on next patch

sn4k3, 2021-02-12:
Implemented

closed 2021-02-12 by sn4k3