sn4k3/UVtools #142: [BUG] UVtools shows pixels missing from layers creating islands even though the original file is fine

Klindel, 2021-02-13:
Layer 356
[image]

Layer 355
[image]

Layer 354
[image]
Here is the same layers in Chitubox
[image]

 The supports are continuous but UVtools says they're not

sn4k3, 2021-02-13:
I see a missing land on between. Maybe chitubox bug? You can't have such precision on the 3D visualizer, but check the 2D layers at chitubox preview

[image]

Daoenti, 2021-02-13:
This is a known bug in Chitubox, it's been reported for versions and versions and they apparently still haven't fixed it.  So this happens when Chitubox slices places that have multiple supports stacked in one spot.  Sometimes it will glitch and end up with a slice that only has three or four individual pixels of the support.  When you run the repair in UVtools it removes those single pixels and leaves you with a blank slice in the support.

It's 100% a Chitubox problem and the reason I switched to Lychee.

sn4k3, 2021-02-13:
UVtools only remove such pixels if they are islands OR if you run noise removal >= 1

Klindel, 2021-02-13:
Thanks! I wasn't aware of the issue, will try something else

closed 2021-02-13 by Klindel