sn4k3/UVtools #143: Error opening the file

Labasso, 2021-02-13:
I generate some PWMX file with the same profile some STL's work but specific others make this error :

Steps to reproduce the behavior:
1. PhotonWorkshop last update same profile for all files
2. Slice to PWMX
3. File => Open
4. There some PWMX works and open but some specific don't

[image]

## Files
I can't like it's a payed one and i can't share it there

- OS: Windows 10 x64

System.AggregateException: One or more errors occurred. (Index was outside the bounds of the array.)
 ---> System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerData.DecodePW0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 555
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerData.Decode(Boolean consumeData) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 415
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass118_0.<DecodeInternally>b__0(Int64 layerIndex) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1226
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException(ICollection exceptions, CancellationToken cancelToken, Exception otherException)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.For(Int64 fromInclusive, Int64 toExclusive, Action`1 body)
   at UVtools.Core.FileFormats.PhotonWorkshopFile.DecodeInternally(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1204
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 1155
   at UVtools.WPF.MainWindow.<>c__DisplayClass90_0.<ProcessFile>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 931

sn4k3, 2021-02-13:
Can you can email me the file so i can debug the problem? But seens like missing layers? I have seen that before with PhotonWorkshop

Labasso, 2021-02-13:
Hello,

Those files don't work :
https://drive.google.com/drive/folders/1FcBk2vEVmzRkTrITmquTdKH8fnKv0tK2?usp=sharing
Could you confirm me that you have acces to my drive ?

Best regards,
Chevalier Saganne

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

sn4k3, 2021-02-13:
Hi, I do, i’m debuging now

https://drive.google.com/drive/folders/1FcBk2vEVmzRkTrITmquTdKH8fnKv0tK2?usp=sharing

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

Labasso, 2021-02-13:
Ok thank you a lot ! What can i do in the futur to prevent this error ?

sn4k3, 2021-02-13:
So if you slice other files sometime works others dont?
Serious i will not use PhotonWorkshop... Even chitubox is better

Labasso, 2021-02-13:
Ok today i'm using prusa for support and I use photon just to slice pre-suported files but ok if photon is that bad even in just slicing yes i will definitely switch !

sn4k3, 2021-02-13:
Problem here is with decoding, maybe they change something, as it exausts the buffer at 64% of the data

sn4k3, 2021-02-13:
I would try same stl on chitubox and see if it opens

Labasso, 2021-02-13:
Yeps i'm trying i never thought that the brand slicer could fail at just slicing !

sn4k3, 2021-02-13:
I dont think it is failing, if you can print the file then the slicing it's ok. Have you tried to print any of this files?

Labasso, 2021-02-13:
nope I every time put them through uvtools. And yes issue solved i just sliced the file and it's showing up in uvtools ! Thank you a lot you save me for this problem

sn4k3, 2021-02-13:
PhotonWorkshop have some problems i would not use it util they fix it.
I also dont recommend chitubox but they are improving.
Lychee or PrusaSlicer is my recommendation

sn4k3, 2021-02-16:
Found the problem, is with AA level set on slicer, if you define as 1 it will work but that will not use AA as well.
Will try to fix on next patch to allow read all levels

sn4k3, 2021-08-17:
Finally, fixed :)

closed 2021-08-17 by sn4k3