sn4k3/UVtools #144: [FEATURE REQUEST] Multiple exposure tests in one print using AA

nvx, 2021-02-13:
## Is your feature request related to a problem? Please describe
When testing a new resin, often you'll need to test a few different exposure times to dial in exactly the best time for your particular resin. With traditional test models, this involves printing the same file over and over again which is quite time consuming.

## Describe the solution you'd like
For printers that support AA, this is achieved on the printer by showing the light for only a % of the time, for example a 20 second layer exposure time, pixels with an AA value of 127 will only have the light on for 10 seconds.

Extending the existing Exposure Time Finder tool to be able to print a number of simultaneous tests side by side using a common value for the base exposure (this gives a nice substrate to be able to hold the simultaneous tests together) and variable exposure times for each individual tests (similar to how the Elephants Foot test does multiple simultaneous tests at once, except in this case they would be joined together).

Being on a common substrate, the parameters that are currently printed in the lower right corner of the exposure test could be simplified to only include the specific per-test settings, while global settings such as layer height and base exposure could be printed larger at the top or bottom, along with user-enterable test. I usually find myself writing in marker on tests so I can remember what resin it is for/etc, having the ability to write some free text would be super handy.

## Additional context
This has been experimented with before for exposure tests:
* https://github.com/Photonsters/Resin-exposure-finder-v2/releases
* https://www.facebook.com/719355768518406/posts/resin-exposure-finder-v2-beta-and-validation-matrix-stl-is-now-online-and-accept/1145550492565596/

The downside of these approaches is they are by design relatively static tests. The flexibility of UVtools to be able to generate on-demand pixel perfect tests with custom parameters is a game changer.

sn4k3, 2021-02-15:
If i understand correctly you want a Dimmer inbuilt on?
[image]

nvx, 2021-02-15:
Yup that sounds about right. Good point on the custom text, that should work nicely.

By not enforcing a common substrate though, do you mean there will be an option for it? If so I'm happy with that, fair enough not forcing it globally if an option is suitable.

I guess the only thing missing from what you said is doing the math to convert the brightness % to an exposure time in seconds. At least on the Anycubic Photon 127/255 will result in a 50% shorter exposure time, so a bit of math can be done to show a second value instead. Maybe an option to switch between AA brightness level and seconds math is a better option if this isn't the case across all printers?

sn4k3, 2021-02-15:
> By not enforcing a common substrate though, do you mean there will be an option for it? If so I'm happy with that, fair enough not forcing it globally if an option is suitable.

The option was been always there, part margin is set to 0 by default which mean parts print glued, if you increase that margin they will be distanced by that value and no longer a commum piece.

> I guess the only thing missing from what you said is doing the math to convert the brightness % to an exposure time in seconds. At least on the Anycubic Photon 127/255 will result in a 50% shorter exposure time, so a bit of math can be done to show a second value instead. 
I can show and print the converted brightness value in seconds as well. but i don't know at which point that is valid, some guys say that the brightness are not linear convertion, but i dont really know, never had done any study. In teory 127.5 is half of your exposure but some say that will not the same as print full white at that time instead. Still i will allow that to show up.

> Maybe an option to switch between AA brightness level and seconds math is a better option if this isn't the case across all printers?

For now i will skip that, since uvtools logic and image seens pixel brightness and not seconds values, also use seconds mean i will have to weak users, because if they have a set to 3s and on brightness field they put 4s i will have to auto fix and tweak values and user may fell something is wrong, i can also show error messages etc but for now lets keep with the simpler logic, brightness values are over thier top settings.
What i can do however is to include a generator where you put the exposure time you want and uvtools append that to the list, sanitized

sn4k3, 2021-02-15:
[image]

sn4k3, 2021-02-15:
[image]

sn4k3, 2021-02-15:
Implemented

closed 2021-02-15 by sn4k3