sn4k3/UVtools #145: [FEATURE REQUEST] Ability to change text in calibration tests from inset to embossed

nvx, 2021-02-13:
## Is your feature request related to a problem? Please describe

A picture is worth a thousand words:
[image]

Both of these models were printed at the same time. I'm not sure if it's due to differences in light levels across the print bed or other voodoo, but sufficiently small holes like used in rendering the text I find sometimes don't show up properly depending on exact exposure settings/etc.

Needless to say it's a bit hard to read when it fills in like that.

## Describe the solution you'd like
Being able to change the text mode from a hole to sticking up (similar to how the alphabet shows on the resin test models) would solve this issue. I'm sure some would like the text as it currently is, so a tickbox to change the text mode from inset to embossed (and when embossed is enabled an option for how high the text should be should appear) would be the best of both worlds.

## Additional context
Another issue with the current mode is that characters like "0" end up printing an unconnected island in the middle of the letter. In my prints the combination of elephants foot and the tendency to morph into one solid blob has meant this is a non-issue in practice, but I imagine printing a sufficiently scaled up model would end up having the middle bit falling out (or remaining stuck to the bed when removing the model). Having the text not protrude through the base layers would solve that issue too.

sn4k3, 2021-02-13:
I could implement that.
I guess you on 4K display?
The parts are so small that text must be too...
But the 2nd part you show there are something wrong, maybe light not uniformal? If you not dim the center that should not happen.
Anyway what i recomend is to save the profile you used to generate the test, or screenshot the result, or reopen the file and check from there, the test number seens to be always visible (1...x)

nvx, 2021-02-15:
nah I'm on a 2k display, but 5.5 inch (original Anycubic Photon). It was the default size for the test pieces go figure. I could make the pieces bigger, but that size was fine for the test really, just the text was a bit iffie.

The OG Photon only has a single UV light and no fancy lens diffuser thing either, so almost certainly non-uniform light. The sliced file looks fine.

sn4k3, 2021-02-15:
Implemented

closed 2021-02-15 by sn4k3