sn4k3/UVtools #146: [BUG] Layer height "permanently" zero when importing from images

femtostar, 2021-02-14:
## Describe the bug
When an image is opened in UVtools, rather than a file from a slicer, it defaults to a layer height of 0.00mm. This creates a one-layer file. The "Adjust layer height" window asks you to clone the layer, which you can do. Once this is done, however, "adjust layer height" cannot change the layer height, as it multiplies the (zero) layer height and still comes out with zero.

## To Reproduce
Steps to reproduce the behavior:
1. Open an image in the latest UVtools
2. Confirm that it has opened as a single layer
3. Clone the layer (Actions > Clone layers) 19 times
4. Confirm that you now have 20 layers
5. Go to Tools > Adjust layer height
6. Drop down "Modifier"
7. Note that all options still result in infinitely-thin layers

## Expected behavior
Adjust layer height, or some other feature, allows the imported 0mm layers to be thickened to, for example, 0.05mm, such that they can be printed.

## Screenshots
All options result in 0mm thick layers
[image]

## Files
Example image, which it imports as expected but is permanently stuck at 0mm thick.
[image]

## System
**Desktop (please complete the following information):**
 - OS: Debian testing GNU/Linux
 - Openbox on X11
 - UVtools 2.4.5

## Additional context
I understand why adjust layer height works by multiples to avoid breaking models, but 0mm thick layers are somewhat of a special case - perhaps an option could be added to set a model-wide arbitrary thickness that 0mm-thick layers will be converted to? Or perhaps layer thickness could be asked in a popup when importing from a 2D format such as an image.

sn4k3, 2021-02-14:
UVtools image files (png, jpeg, etc) are made to work only with a single image. While you can clone, import, process etc and it actually works to build a stack of images, if you save (File -> Save) only the first layer will be saved and others will not take into consideration. However the File -> Extract can export all stack to a foder.
I can unlock the height and set to 0.01mm as default, but anyway the Adjust layer height for a stack of images are just usless as that only remove or duplicate images.
Can you explain your use case and what are you trying to do?

femtostar, 2021-02-15:
I'm trying to generate a file that will print a single-layer-thick raster image onto a substrate material mounted on the print bed - kind of an unusual use of an SLA printer but I want to experiment with it. I want to have pixel-by-pixel control over the printer's display so that I can make small features without built-in anti-aliasing, and then introduce anti-aliasing to see how it affects the produced layer, rather than slicing a 3D model to layers and relying on PrusaSlicer to handle making rasters of the layers - this is why I'm trying to import directly from images. The fact that it only handles a single layer properly is okay - I'm only trying to print a single layer. The reason I duplicated it is only because Adjust layer height said I had to.

sn4k3, 2021-02-15:
You can have a dummy file like .ctb or zip or sl1, use the Actions -> Import layers to rebuild the file, there you can clean the file to start with only your 1 image. But i will set layer height to be 0.01mm by default on images.
Adjust layer height says to duplicate layers when is not able to have valid multipliers or divisors, that is universal check independent from the format.

femtostar, 2021-02-15:
What if I want a layer thicker than 0.01mm (0.03 or so) but I want it to be cured in one layer rather than duplicating the layer three times? Would there be some way to have like a setting or a popup, or even just a command like arg like `--image-default-thickness 0.05`?

sn4k3, 2021-02-15:
Images have no thickness, still UVtools will accept to modify the thinkness using the tools. What file format your printer uses?

femtostar, 2021-02-15:
I know the image has no thickness, what I meant is I need the ability to set the thickness within UVtools so that's fine. I use a Creality LD-002H which takes CTBs.

sn4k3, 2021-02-15:
Ok so in that way you can pre-slice a ctb file with the thickness you want and all the settings.
Then use UVtools to replace all content with your single image, note that the imported image must have same resolution as printer
[image]

femtostar, 2021-02-15:
Hi there!

Just tried this, seems like a good idea, but it crashes for me.
```
Unhandled exception. Emgu.CV.Util.CvException: OpenCV: 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
Aborted
```

sn4k3, 2021-02-15:
Maybe the image you import have diferent size from printer resolution?
Send prints of your UVtools window with file loaded and your import dialog

sn4k3, 2021-02-15:
Ok nvm, i found that bug, i will also fix it next patch, but that is also from images that dont have same resolution, if you use same resolution image it will import fine

femtostar, 2021-02-15:
Ah, you're right, size was wrong. I had the image I was importing at 2560x1620 rather than 1620x2560. Might be good to give a clearer error in that case but you're right, that's my mistake.

sn4k3, 2021-02-15:
Fixed

closed 2021-02-15 by sn4k3