sn4k3/UVtools #147: [FEATURE REQUEST] Create a project from a bmp/png etc...

SteveDC2, 2021-02-15:
I am experimenting with using my printer to expose PCBs and the ability to simply load a bmp/png and then "convert to" a CTB project would be REALLY useful. An extension to this could be to import multiple images automatically and use each one for a layer, then create a project from that. This could open up the ability for other methods of generating sliced data in a simple, well understood format (i.e. bmp). This feature is ALMOST present, but seems to only be possible within an existing full project and the image size has to be the same is the current project.

Currently if I open a bmp image the option to "convert to" disappears. I understand why since there is no printer specific information but allowing that to be entered (or be able to load printer profiles) would be nice.

The best option I can think of at the moment is to create a project in Chitubox that has a small model that is 0.05mm high (or whatever the layer height is so that only 1 layer is generated) then use "Actions->Import layers" in UVTools to bring in my pre-generated bmp before re-saving the updated project.

Side note : UVTools is absolutely fantastic. It is an EXCELLENT tool for cleaning up the mess that Chitubox sometimes creates !! Keep up the great work.

sn4k3, 2021-02-15:
That is already possible, please see #146 

> This feature is ALMOST present, but seems to only be possible within an existing full project and the image size has to be the same is the current project.

That's wrong, while UVtools says you must have same resolution, you can import other resolution images IF volume bounds are inside printer resolution, otherwise will not be usefull and are discareded.
But current import tool have a bug that i must fix. But for example you can try out the Replace, if images have same resolution its fine, if not i need to fix that on next patch.

My recommendation is that, use a pre fabricated file and replace all layers with your data

sn4k3, 2021-02-15:
Fixed importation tool, use that instead

closed 2021-02-15 by sn4k3