sn4k3/UVtools #148: [FEATURE REQUEST] Specify location of Actions->Import Layers

SteveDC2, 2021-02-16:
First, thanks for getting to my other issue so fast !!! Import Layers is a LOT more stable now. Very much appreciated.

For my PCB exposing it would be useful to be able to put the imported image in any of the 9 locations i.e. top left, top center, top right, right center, bottom right, bottom center, bottom left, left center or screen center.

This allows much simpler alignment of the PCB to the image.

It is also important to use the source image rectangle bounds and not the active image bounds within the file. 

Just donated too. UVTools is one of the most useful tools I have used in a while. Thanks for working on it.

sn4k3, 2021-02-16:
UVtools always put imported image on center or where it got space for depending on the type you are using.
After importing you can use the Move tool which also allow you to move or copy a ROI and select advanced positioning

[image]

sn4k3, 2021-02-21:
For now i'm not replicating Move inside Import, use move tool

closed 2021-02-21 by sn4k3

SteveDC2, 2021-02-22:
OK, not sure what happened but I posted a whole response to this that seems to have disappeared !!!

The issue with the above is that the full bounds of the image imported are lost. This means that the image is moved to the wrong location.

For example, if the LCD size is 1000x600 and if the image imported is say 300x300 pixels but only has white pixels in the top right 10x10 then this 10x10 set of pixels will be moved completely to the edges of the LCD, which is not useful for this use case.

It would REALLY help if the region moved had bounds that matched the imported image. I understand that once the image is merged the bounds are lost since this is not an object manipulation tool, but an image manipulation tool so this is why it really needs to be in the import flow and not the subsequent image manipulation flow.

My only solution I can think of is to include a single pixel in each corner of the imported image so that the bounds are retained.

Hope this makes sense.

As I say, not critical and not a 'normal' use case, but would certainly be useful for me but I have a work around (dots in the corners).

I might look at the source code and add it myself though.

Regards,
Steve

sn4k3, 2021-02-22:
> For example, if the LCD size is 1000x600 and if the image imported is say 300x300 pixels but only has white pixels in the top right 10x10 then this 10x10 set of pixels will be moved completely to the edges of the LCD, which is not useful for this use case.

You can replicate that?
Because right now UVtools grab the white area and paste in center most at printer resolution, so that 10x10 pixels would be placed at: Resolution / 2 - 10 / 2 = Center

SteveDC2, 2021-02-22:
Agreed UV tools centers the active pixels, but that isn't what I need :) I need it to bound the ENTIRE loaded image so I can move the entire original image, not just the active pixels.

I need the entire original 300x300 image to be placed at the edge/corner.
For example, reading in the 300x300 image that has 10x10 active pixels in the top right corner of the source image, I want to be able to load that image in to the bottom left LCD corner and end up with the 10x10 pixels located at x = 300-10 and y = 600 - 300 + 10 (i.e. the original 300x300 lower left is located at the LCD lower left completely regardless of the active pixels.

All of this activity allows me to consistently place the PCB in the corner of the LCD. The location is therefore very accurate. Aligning the PCB to the center of the LCD is impossible to do repeatably without a jig.

Anyhow, as I said previously this is not a big issue and I have a solution that gets me what I need (1 pixel in each corner of the source image) so don't worry too much abut this. I am a corner case for UVTools, and I understand that :)

BR,
Steve

sn4k3, 2021-02-22:
I guess that can be made with Move tool but you want it on import to skip a step right?

SteveDC2, 2021-02-22:
If I add the corner markers then I can do the move correctly after loading, so that is my solution for the moment.

If I don't add the corner markers then it is impossible to move the image after importing since the original size of the ENTIRE imported image is lost.

For PCB exposure it is critical to keep the entire image relationships.

Here is a pathological example...
I have a PCB that is 2 inches wide by 1inch tall. I place the blank unexposed PCB in the bottom left corner of the LCD. I then have a PCB design that, for this example, is simply a 1/4" x 1/4" logo that should appear at the top/center of the PCB. I create a bitmap for the PCB at the correct resolution to yield a 3" x 2" bitmap (the size of my PCB). The bitmap has the logo at the correct center top of the image. 

When I load this bitmap into  UTTools the logo will end up centered on the LCD and a bounding box created ONLY around the 1/4" x 1/4" logo. At this point it is impossible to move it exactly to the correct position on the PCB since there is no longer any information about the original 3" x 2" image.

[image]

sn4k3, 2021-02-23:
I see, so you not only need only the move but you also need to keep image integrity, because right now UVtools always go for white area to allow larger images to fit in, even if they are smaller, overall uvtools will pick the bounds. Only images with same resolution it just replace the image without crop or move, so that can be a solution too, if you draw at printer resolution images.
I can in the future include a option to go for bounds or whole image and allow to use a move inside the import tool.
Your hack will also work, by doing a frame or a position pin.

You can also make use of UVtools scripting to make your own "tool" to work as you need: https://github.com/sn4k3/UVtools/tree/master/UVtools.Scripts

SteveDC2, 2021-02-23:
Exactly :)

Like I say, not a huge issue and I am doing something UVTools was/is not designed for, but it is REALLY close to what I need :)

I have a work-around so no rush.

As always, your work, effort and time are very much appreciated. Thanks :)

BR,
Steve