sn4k3/UVtools #150: [BUG] Bug in time calculation for VLH + CTRL Z crash

Nume1977, 2021-02-17:
## Describe the bug
If i make a variable layer height of minimum 0.1mm / max 0.2mm with linear steps of 1s for the itineration's, the resulting time at the end does not change, but the fact is I can do the print it in about 5H (-3H)

[image]

Also, executing "undo" after vlh will crash UVTools :P

sn4k3, 2021-02-17:
I will check that.
Also Undo this operation is very aggressive

sn4k3, 2021-02-17:
Works for me, undo as well.
[image]

What file format you are using?

Nume1977, 2021-02-17:
I am using PrusaSlicer 2.3.0 (with the Nova3d printer installed from UVtools) to save a sl1 file for nova3D (cws).
I have just sliced a fresh stl, using the printer profile and universal 0.035 and always have the same result when doing VLH on UVTools...

Maybe it has something to do with the file type (CWS)?

[image]

sn4k3, 2021-02-17:
Never use more than 2 decimals for layer height! 0.xx max!
I dont have universal 0.035 profiles... please check that

Nume1977, 2021-02-17:
Sorry my typo, I meant 0.03mm!

sn4k3, 2021-02-17:
If you have them by a old version, please delete them

[image]

sn4k3, 2021-02-17:
Ok just tried with a cws, still work for me, both print time and undo...
Can you send me your file?

Nume1977, 2021-02-17:
Weird!
I am sending a 0.2mm sliced SL1 (0.05 is over 100Mb!)
[FalconWith_Landing_Gear_0.2mm.sl1.zip](https://github.com/sn4k3/UVtools/files/5997972/FalconWith_Landing_Gear_0.2mm.sl1.zip)

sn4k3, 2021-02-17:
ok i found the problem, is due the auto convertion from sl1 to cws. After open the sl1 if you refresh (Ctrl + F5) it will work

Nume1977, 2021-02-17:
CTRL + F5 fixes the issue!

sn4k3, 2021-02-18:
Fixed

closed 2021-02-18 by sn4k3