sn4k3/UVtools #151: [BUG] Print time calculation

Nume1977, 2021-02-17:
## Describe the bug
After changing "Exposure time" in print parameters, the Print time remains the same of the original file (never gets recalculated), and there is a bug on edit box, it displays the original time.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Edit print parameters'
2. Click on 'Exposure time' and change value
3. Accept
4. Goto step 1 to see 'Exposure time', it returned to the original value

sn4k3, 2021-02-17:
Works for me xD
Please complete with file format

Nume1977, 2021-02-17:
LOL!
Comecei pela versão portable (Zip), mas depois tive que apagar a diretoria e instalar a 2.4.0 através do msi e a partir daí só usei o auto-update... Será configs das versões antigas que não foram apagados!? Onde posso apagar?
Acho que todos os problemas vêm daí ...

sn4k3, 2021-02-17:
Keep talks in english only.
MSI have problems, i recommend to uninstall first then install.
The configuration files have nothing to do with this problem, again the problem should come from the auto converted file, CTRL + F5 should also solve that

Nume1977, 2021-02-17:
CTRL + F5 fixes the issue!

sn4k3, 2021-02-17:
After auto convertion always use the .cws, if you reopen the .sl1 by mistake the .cws will be overwritten

Nume1977, 2021-02-17:
Will do, thanks mate!

closed 2021-02-17 by Nume1977