sn4k3/UVtools #152: [BUG] Morph window resizes wrong amount when kernel options are expanded

tofof, 2021-02-18:
## Describe the bug
When the kernel options are expanded open in the morph window, the morph window resizes to accomodate the extra stuff. However, it doesn't get enough larger, and the OK / Cancel buttons end up mostly cut off.

## To Reproduce
Steps to reproduce the behavior:
1. Morph window
2. Click on Kernel options - expand

## Expected behavior
The morph window just needs to get another 50ish px larger than it currently does.

## System
* OS: Win 10 Pro x64 1803
* UVtools 2.4.8

Tons of updates recently! Keep up the great work but don't work too hard!

sn4k3, 2021-02-18:
I guess this is on scaling > 100% ?
This is not UVtools related, but the UI framework have tons of bugs in that way, lets wait and hope they fix most of then on coming releases

tofof, 2021-02-18:
No, normal 100% scaling on a 1920x1080.

The setting "take into account the screen scale factor to limit the dialogs windows maximum size" has no effect on this behavior.

[image]

tofof, 2021-02-18:
Possibly related - in the settings dialog, the miscellaneous tab can't be scrolled all the way to the bottom unless I click into the 'issues' tab first, which gets the 'save/cancel' bar to show up and then I can scroll all the way down.

[image]

sn4k3, 2021-02-18:
Thats strange, because on my same resolution as you shows all 
[image]

Post screenshot of F1 dialog

tofof, 2021-02-18:
[image]

sn4k3, 2021-02-18:
Ok nothing to flag on that about box. Its just strange why works on my but on others dont... My even work with 175% scale without cuts

tofof, 2021-02-18:
Yup, weird. I believe you that it's a problem for the ui framework, though, so I'm going to close this.

closed 2021-02-18 by tofof

sn4k3, 2021-02-18:
It is, i already have to apply some hacks for windows to show up "correctly", i will not keep wasting time on this since this must be fixed on core and should be a painless thing to deal with