sn4k3/UVtools #153: [FEATURE REQUEST] Elephants Foot Test - Chamfer for morph-erode

MrCythos, 2021-02-20:
Hi,

I've been testing the elephants foot calibration function with a few printers and I recently got an Anycubic Photon Mono X. Unfortunately, it has transition layers tied into the firmware which is not user modifiable. These transition layers break the elephants foot calibration, however, I am able to use the chamfer function in the morph-erode tool to deal with these transition layers within UV-tools.

Would it be possible to add the chamfer function that's in the normal morph-erode tool into the elephants foot calibration?

sn4k3, 2021-02-21:
> These transition layers break the elephants foot calibration

I guess transition stop at a point? You can simply increase the bottom layers.
How add a chamfer undo the transition layers?

MrCythos, 2021-02-23:
My apologies for the late response. Anycubic has added 10 transition layers on top of the number of base layers, which scale linearly from the base exposure to the normal layer exposure. Unfortunately, this can't be resolved by increasing or decreasing the number of base layers. In speaking with Anycubic engineers, they have so far refused user control of these transition layers.

Due to those transition layers, the elephants foot calibration test doesn't work with the Anycubic Mono X as once the erosion is stops after the base layers, the transition layers are again overexposed and cause the test prints to partially fuse with one another.

For my prints, I've been using the morph-erode function to erode the base layers (n-1) of the part. Then to compensate for the default transition layers, I add a chamfer between layers n to n+9 from to erode from the erode iteration that worked for the base layers, to 1. 

[image]

You can see an example of the profiles I'm currently using for the liqcreate deep blue resin, which has 2 base layers. I erode the  first layer (n-1) by 6 iterations, then I use a second profile from layers 1-11 (n+10) which chamfer from 6 iterations to 1, at which point the layers print normally.

It would be great if this chamfer function could be present in the elephants food calibration test so that I could use that test with these Anycubic printers to calibrate different resins.

FricoRico, 2021-03-02:
I heard somewhere that the RERF files provided by Anycubic do not have these transition layers and go straight from bottom exposure to normal exposure. Perhaps you can validate this claim by taking a look at their GCODE? You can find the files at their website:

https://www.anycubic.com/products/photon-mono-x-resin-printer

FricoRico, 2021-03-11:
It seems my latest statement might be inorrect. As the manual states that filenames called R_E_R_F are beeing treated with special firmware settings.

[image]

sn4k3, 2021-03-11:
Dirty hack, i can't understand this enforcements... They will only learn/listen to others when people spot buying thier products

X3msnake, 2021-03-11:
nope not even then

11/03/2021 à(s) 15:37:

closed 2021-04-30 by sn4k3