sn4k3/UVtools #156: [BUG]

nicexe, 2021-02-21:
## Describe the bug
Converting for Photon (the original model) results in a file not readable by the firmware. The converted file does not show up on the file browser.

## To Reproduce
Steps to reproduce the behavior:
1. Slice a file with prusa slicer and save it.
2. Convert the sl1 file using UVTools to photon and save to a USB drive.
3. Insert the USB drive to photon.
4. Observe the file missing from the file browser.

## Expected behavior
The converted file should be seen in the file browser on the photon.

## System
**Desktop:**
 - OS: Windows 10 Pro x64 v20H2]

sn4k3, 2021-02-21:
1. Try with another pen drive, formated to FAT32.
2. Check if you have last firmware on printer

nicexe, 2021-02-21:
1. The USB drive is fine. All other sliced files are shown and have the correct extension (.photon). The converted file has a different extensions and this is why it isn't shown in the printer's file browser.
2. I'm on v4.2.18 since the latest (v4.2.19) is a bit buggy in some aspects.

sn4k3, 2021-02-21:
What file extension does it have?

nicexe, 2021-02-21:
PWS

sn4k3, 2021-02-21:
Then what profile you are using on prusa slicer? Photon S?

nicexe, 2021-02-21:
Anycubic Photon on Prusa Slicer and converting the resulting SL1 to Photon / Photon S using UVtools

sn4k3, 2021-02-21:
Maybe PWS is not supported by that firmware version.
The new profiles from UVtools on PrusaSlicer auto convert files, if you use Photon it will auto convert to ctb.
Please update profiles from manager

ComteNiark, 2021-03-05:
In the latest release, .photon files don't slice/convert properly. They end up as a corrupted mess: 
[image]

Original Photon machines can't read .pws, so users are stuck using this file format.

This is fixed by closing Prusaslicer and updating the printer profile from UVTools. I assume the "notes" formatting changed in a recent version?

sn4k3, 2021-03-05:
You have to keep profiles updated with my releases, that seen a very old profile where xy where flipped.
i use PWS format because is what PhotonWorkShop output for Photon printer. Maybe it will require a firmware update for the old machines. But you can change in printer notes which file format it auto convert to

ComteNiark, 2021-03-05:
Yes, I figured as much, and not updating for this long was a mistake on my part to begin with. I just wanted to point out this possible solution to the OP, since I encountered a similar problem.

For your information though since I don't think you own one of these machines, the .photon/.pws incompatibility issue comes from the fact that Anycubic actually sold two different machines as "Photon". 
The original photon can only read .photon files on the latest firmware (2.19), whereas another machine, which Anycubic calls "Photon Upgraded", nicknamed the "Fauxton" by the community can read .pws because it is essentially a Photon S (upgraded model) motherboard in an OG photon enclosure. Both drivers are even on the same download page.
The thing is that when you buy a Photon, you have no way to know if you're going to receive a Photon or a Fauxton, it's a bit of a dice roll, since as far as Anycubic's shop is concerned, they're both the same product. I would believe that they updated their photon workshop software to cover the Fauxton since it's the more recent product, and that's why it produces .pws files? But who knows with that company...

sn4k3, 2021-03-05:
Yes that is a mess... i'm using ctb as default for photon printer, what should i use  better? Does it even read ctb at all?

ComteNiark, 2021-03-05:
AFAIK, no printer reads .ctb. It's a file format specific to the Chitubox software, though I couldn't tell you what it's used for. Maybe chitubox slices to .ctb and then converts to the chosen format, as a way to keep their operations clean and separate?

For the photon printer though, you should use .photon as the default. All Photons can read that format, it's just that some updated models can also read .pws. The users can figure out what printer they have and whether or not they should swap to .pws on their own.

sn4k3, 2021-03-06:
most printers with chitu board now use ctb, it is the evolution of cbddlp, eg: epax, phrozen, elegoo use that format for the new printers

sn4k3, 2021-03-11:
OP was using the wrong format, use .photon

closed 2021-03-11 by sn4k3