sn4k3/UVtools #161: [FEATURE REQUEST] Pixel Formula Calculator

ReinhardC, 2021-03-06:
Hello, 

my suggestion is for a pixel formula calculator. There would be a new entry "Pixel Formula Calculator" under the Tools menu which gives you a dialog similar to that of "Arithmetic".

In a formula field, the user can enter a formula that is calculated for each pixel. The result of the formula is set as the new brightness value for the current pixel. 

The following variables ready to use the formula:

```
Xmax: resolution in X direction
Ymax: resolution in Y direction
X: pixel X coordinate of the current pixel
Y: pixel Y coordinate of the current pixel
```

And a function

`B(x,y): get the pixel brightness for the pixel at x, y`

The simplest example `B(X,Y)` would return the brightess of the current pixel (no effect). Some math functions would be provided like `abs(x), sin(x), atan2(y,x), sqrt(x), pow(x, y), min(a,b), max(a.b)` etc.

And a simple conditional with in the usual java/c++ form `[condition] ? [exp_if_true] : [exp_if_false]`

Some examples: 

* set brightness to 50%: `B(X,Y)/2`
* apply a AA grey value of 50% for a sliced files that had a 0% grey value: `B(X,Y) == 0 ? 0 : 0.5+B(X,Y)/2`
* appy a gamma to the brightness: `pow(B(X,Y), 1/2.2)`

there could be calculations usiung neighboring pixels using for example `(B(X+1, Y-1)+B(X-1, Y+1))/2`

Some of these example are useless, some can be reproduced using the other tools, but maybe as a whole this feature would be useful

sn4k3, 2021-03-06:
This feature looks like a good use case for UVtools Scripts: https://github.com/sn4k3/UVtools/tree/master/UVtools.Scripts

ReinhardC, 2021-03-06:
Thanks, it looks a little intimidating but I'll try

sn4k3, 2021-03-06:
> Thanks, it looks a little intimidating but I'll try

It's not very hard to learn, when you get in it will be peanuts to code

sn4k3, 2021-03-11:
Work for scripts

closed 2021-03-11 by sn4k3