sn4k3/UVtools #166: [BUG] Attempting to load in a file from a USB drive rather than disk caushes the app to crash.

jamestrotter, 2021-03-13:
## Describe the bug
I slice all of my files directly to the thumb drive I plug in to my printer. Attempting to view these files directly from the USB drive in UVtools causes the application to crash. Moving the files to my PC and opening them is successful.

## To Reproduce
Steps to reproduce the behavior:

1. Export a sliced file to USB stick
2. Open UV tools
3. Attempt to load the file from the USB drive

## Expected behavior
The file the open the same regardless of the drive type

## System
**Desktop (please complete the following information):**
 - OS: Windows 10 x64

sn4k3, 2021-03-13:
Others have reported this, but i can't replicate on my pc even with my worst pen drive.
Can you send me a file that fails to load? 
And pen specs and partition type?

jamestrotter, 2021-03-13:
[Whole_Pelican.zip](https://github.com/sn4k3/UVtools/files/6135694/Whole_Pelican.zip) is one of the files that crashed on the thumb drive.

Can't speak for specs other than it being 16GB and probably USB2 since its the drive that came with the printer.. so probably on the cheapest end of the spectrum.

Is there a guide for getting the app to build? I was going to take a look but I was hitting issues with getting it running. My first issue was that I couldn't target .NET 5 but installing the VS preview seems to resolve that. Now when I try and build the WPF project (I assume that's the one to build - the GUI project tells me the target versions are mismatched) I get
`FileNotFoundException: The resource avares://UVtools/Assets/Icons/geometry-16x16.png could not be found.`
on launch.

sn4k3, 2021-03-13:
Ok i will take a look, the partition type is FAT32?

You require to install .NET 5 SDK as this not ship yet on windows nor VS stable.
You just need to open the UVtools.sln, it should compile out of the box and run.
the UVtools.GUI and Installers are not required and you can unload them (They are already from .sln)

> FileNotFoundException: The resource avares://UVtools/Assets/Icons/geometry-16x16.png could not be found.
That file exists for sure, but that path seens incorrect it should be avares://Assets/Icons/geometry-16x16.png

sn4k3, 2021-03-19:
Try with v2.7.0, load a file from pen, it will maybe generate a error log at UVtools settings folder, report error back (Help menu - Open settings folder)

sn4k3, 2021-09-21:
Is this still a issue?

sn4k3, 2021-10-29:
Can't replicate this, closing util new evidence

closed 2021-10-29 by sn4k3