sn4k3/UVtools #169: [BUG] Export to Anycubic Photon 0 pw0 format results in noisy unreadable file

RobotSpaceDad, 2021-03-17:
## Describe the bug
When exporting to pw0 I am getting a noisy looking file that is unreadable on my printer. Every layer is broken in this manner.  My sl1 files are coming from PrussiaSlicer and don't exhibit any signs of issues when open as an sl1 in prussia or UVTools, this only happens upon exporting.

[image]

## To Reproduce
Steps to reproduce the behavior:
1. Open a sl1 file in UVTools.  
2. Convert to Photon Zero (PW0)
3. Open output in UVTools
4. See error

Normally I would expect the converted file to look like the original in the preview.  There are no apparent error messages on export.  I've had this across a few files now.

## Files
STL, sl1 and PW0 here: https://drive.google.com/drive/folders/1uGrJ9DbmdhoNCdyWph9UJXaAJOZJo5g6?usp=sharing

## System
**Desktop (please complete the following information):**
 - OS: Microsoft Windows 10 Pro Version 10.0.18363 Build 18363

Let me know if you need anything further
  
Thanks for this tool btw, I think this work you're doing here is really cool :)

sn4k3, 2021-03-17:
It works for me, i think you have an old profile of Zero? Or you did your own?
Go to UVtools and update the printer with Profile Manager

RobotSpaceDad, 2021-03-17:
Yep! That must have been the issue,  thank you very much!  You're a king among men :)

closed 2021-03-17 by RobotSpaceDad