sn4k3/UVtools #172: BUG? Script Dimming issue

ememolero, 2021-03-22:
Hello!

I have started using UV tools scripting to automate tasks but the dimming does not seem to respond to change of brightness or change of pattern. I have tried everything, I even tried to generate an matrix to see if I took it as a parameter. I have read the .cs that manages it trying to understand how it works and it gives me the impression that it generates its own matrix ignoring (chessboard) any other parameter.

Thank you!!

Mario

ememolero, 2021-03-22:
PS: I have followed all the steps carefully, installed Visual Code and have easily managed to use a morph script guiding me through the example without problem.

sn4k3, 2021-03-22:
The **OperationPixelDimming** is a bit different from others, the **Brightness**, **InfillGenThickness**, **InfillGenSpacing** are only for displaying, save the values for profiles and generate the text which is parsed latter but they will not be used on **Execute**, what you have to do is set the **PatternText** on your script and there you set the values directly with the brightness. Look at **GeneratePixelDimming()** function sample.
You can set Pattern matrix directly but if you want to use the PatternText you need to call the **Validator** which builds the matrix from text.

Example:

```C#
op.PatternText = "0 255\n255 0"
op.Validate();
op.Execute();
```

As good pratice you can always call .Validate() at every operation before execute and do some if validation to not execute in case errors, see how UVtools handles an operation: https://github.com/sn4k3/UVtools/blob/3f838001c7fe5d67dadfd3d70e7ea012f99fabe7/UVtools.GUI/Controls/CtrlToolWindowContent.cs#L205

ememolero, 2021-03-23:
Wow! Thanks for the quick response and great explanation. Now everything makes much more sense, my programming capacity is still limited but I have understood very well the way forward. Thanks a lot!! I will write a progress of my progress.
Best Regards,

Mario. #

closed 2021-03-23 by sn4k3

sn4k3, 2021-03-23:
Use this before call execute:

```C#
$validation = $op.Validate()
if(![string]::IsNullOrEmpty($validation)) {
    Write-Error $validation
    return
}
```

ememolero, 2021-03-23:
Ok. Thank you. I managed to correctly generate the pixel dimming like a charm but certainly validate information only answered "true" if everything went right. I will test that.

 I have combined 3 operations without problem(raft relief,new matrix for a morph op, and pixel dimming), congratulations for the great work,it is a powerful tool and it is saving me a lot of time.

Another issue, this time is a real issue?😅; When I apply Pixel dimming to a wall (Ex: 4 pixel, Solid, 127 Brightness) the previous Anti-Aliasing from Chitubox is "corrupted" by the op. Some brighter pixels disappears or change their bright value to something wrong and the Anti-Aliasing effect is lost. I'm trying to do the same without the AA from Chitubox and apply it after with UV tools wirh blur op but the AA effect produced isn't right for me, I think.
Many thanks!
Mario

sn4k3, 2021-03-23:
Yeah pixeldimming was made to inner mass, then i adapted for walls, i will check