sn4k3/UVtools #177: Issue detection breaks when Islands [YES] + Overhangs [NO]

tofof, 2021-04-01:
## Description
When settings are to compute islands but not overhangs, issue searches do not execute at all.

## To Reproduce
Steps to reproduce the behavior:
1. Settings -> Issues -> Islands [:heavy_check_mark:], Overhangs [&nbsp;&nbsp;&nbsp;]
2. Save, restart, detect issues (automatically or manually depending on other settings). Nothing happens.

## Other Information
Setting `Enhance island detection with an combined overhang check` does not influence bugged behavior.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(Trivial / Localization - that setting should use "a", not "an")

Islands [&nbsp;&nbsp;&nbsp;] / Overhangs [:heavy_check_mark:] works as expected. 
Islands [&nbsp;&nbsp;&nbsp;] / Overhangs [&nbsp;&nbsp;&nbsp;] works as expected. 
Islands [:heavy_check_mark:] / Overhangs [:heavy_check_mark:] works as expected.

## Expected behavior
Islands check functions independently from overhang check as in prior versions.
(Known good behavior in 2.4.8).

## System
* OS: Win 10 Pro x64 20H2 19042.867
* UVtools 2.7.0, 2.7.2

sn4k3, 2021-04-01:
Works for me on the DEV version
[image]

I will check on that version to see if is fixed now

tofof, 2021-04-01:
> EDIT: To bug happens, touching bounds and empty layer must also be disabled...
FWIW, empty layer setting doesn't change the behavior for me, but you're right that touching bounds must be disabled.

sn4k3, 2021-04-01:
> FWIW, empty layer setting doesn't change the behavior for me, but you're right that touching bounds must be disabled.

Yep, a lacking of parentheses on the spare cycle condition. WIll be fine on next patch.
Thanks for reporting

closed 2021-04-01 by tofof