sn4k3/UVtools #179: [BUG] Most of hollow model detected as resin trap with latest update

eberkain, 2021-04-03:
## Describe the bug
I have been printing busts for the past week+ this is number 20 or so.  Its hollow with a drain hole in the bottom of the base.  With this model when I run the check for errors, I get the base as hollow up to the layer where the base cavity merges with the model cavity, and there it flags everything else as a resin trap. The only thing that has changed since the 3 I sliced yesterday was I updated to the latest version of uv tools. 

This is the file. https://drive.google.com/file/d/1an_7k-xo_CxJASHCmlmumX6bOeZ6SDjP/view?usp=sharing

This is a desktop capture of scrolling through the layers before checking for errors.  https://youtu.be/SJ6UZtYNKIU

This is a desktop capture to show what I am seeing after checking for errors. https://youtu.be/RjBsNAdPuhA

## System
Windows 10 Home 19041.867 64bit
Windows Feature Experience Pack 120.2212.551.0
Ryzen 7 3800X
32GB Ram

eberkain, 2021-04-03:
I went and re-exported one of the models I sliced yesterday, that is very similar to this, and it does not have the same issue. I'm a bit puzzled why this one is acting strangely. I will see if I can reproduce the error with another model.

sn4k3, 2021-04-03:
Yes, i can tell resin traps still have some flaws, i need to review the code and get a way to cover all paths.
I will try to compile a tree of outcomes to fix this problem. Right now it depends on model complexity, paths that cross will often cause this

sn4k3, 2021-04-04:
Should be fixed now

closed 2021-04-04 by sn4k3