sn4k3/UVtools #180: [BUG]

kornshadow097, 2021-04-05:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
A clear and concise description of what the bug is.
i opened a file after updating. all sorts of pop up errors come up saying it couldnt be saved.
then it says my printer resolution doesnt match...but it did before  i updated uvtools..
## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots
If applicable, add screenshots to help explain your problem. Never hide information, full GUI/window screenshot is always a must!

## Files
Were possible attach the problematic file, both source and output file if applicable.

## System
**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 x64 v2004]
  
## Exceptions Log file
In some unexpected exceptions errors will generate and save a log file under appfolder/Logs/AppErrors.
If present you can attach the log file here. Check file name against date/time to be sure it's related to your problem.

## Additional context
Add any other context about the problem here.

sn4k3, 2021-04-05:
You need to take more care when opening a issue (No title, lot of noise, lack or information) 
Templates are to be filled or remove the sections you dont need.

UVtools now check some potential problems on file, that one of them.
What is your printer?

kornshadow097, 2021-04-06:
printer is a nova3d bene4 mono.

first time i used the previous software i had no issues. this update even after clean install it pops up something saying that my machine resolution and file dont match. but i couldnt find any way to set the specs of the file in the software.. not to mention this issue didnt exist before updating. "file and later resolution mistmatch" followed by an error your software wont let me copy paste... i can highlight it but i cant ... and i dont have time to upload the screenshot and post a link

sn4k3, 2021-04-06:
What slicer do you use? I tried my profiles with PrusaSlicer and it works without error.
You can easy copy error messages with the right click of mouse and click on "Copy"

kornshadow097, 2021-04-06:
I mentioned the answer to your question in my. Last reply. I use lychee
slicer. V3.3.2 IIRC

AS FAR as the error I also noted the title in my last reply. Do you need
the actual info of the error?

I tried to copy it but nothing copied to the clipboard. I can try again
later.

But I uninatalled uvtools and reinstalled the old version and now the error
carried over to the old version. Likely due to files being left behind with
the error.

sn4k3, 2021-04-07:
Ok i just check the Lychee configuration, and thier configurations for the printer are wrong.
[image]

Height must be: 1566px and not 1568px, it must be in multiples of 3
Also i dont know why they flip width with height but that values are fliped. However on the configuration file they are on correct orientation.

For this printers that resolution is just informative, but uvtools use it with some tools when you generate images.
That message box is a warning you can just click ok and continue, but i recommend to fix the resolution. The NovaMaker also use that wrong value, but if you inspect the image you will notice i'm right:

[image]

522 x 3 = 1566!

sn4k3, 2021-04-07:
I found an bug present on cws format when using lychee slicer files.
Please dont use uvtools on those cws until new patch coming out. But still, the resolution problem i said before, must be addressed by you

kornshadow097, 2021-04-07:
Interesting find, I'll edit it.

Lychee only let's me change the "volume" field but the screen field is
greyed out.

It's only for pro users

sn4k3, 2021-04-07:
If you only use the repair function, pixel editor and some tools that are not calibration it will be fine and you can ignore that warning.
If you want to fix you can always use Tool - Change resolution and fix that value.
But its bad to see the oficial slicer with the wrong value, lychee is following original slicer settings but in the end they do images in the correct resolution after all.

kornshadow097, 2021-04-07:
Ok

I tried just using repair for supports but when I exported the file for
bene4 mono and used the file... My printer acted weird.. The uv and lifting
were out of sync and the layer time and pauses were all wrong as well. I
ended up with a hard layer on my fep and floaters.

Total mess.

sn4k3, 2021-04-07:
You can send me your file so i can inspect to see if commands are right or if have any problem

closed 2021-04-23 by sn4k3