sn4k3/UVtools #183: [FEATURE REQUEST] Could a logic change to exposure time finder generation dramatically cut print time?

jngeist, 2021-04-12:
Also look at: https://github.com/sn4k3/UVtools/discussions/categories/ideas

<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Is your feature request related to a problem? Please describe

The exposure time finder with multiple exposure times is a fantastic idea, but currently the way UVTools generates slices means that print time for 6 exposure times is effectively the same as running six separate prints. It *seems like* a slight change to logic could make that process much, much faster. 

## Describe the solution you'd like

For the moment, assume I'm generating an exposure tester with everything set the same except normal exposure time. My exposure table has entries for a 6, 6.5, and 7 second normal exposure variants (all other things equal). Let's call the 6 second exposure "A", the 6.5 "B", and the 7.0 "C." The current logic generates as follows:

Layer 1: A, 6.0 seconds
Layer 2: B, 6.5 seconds
Layer 3: C, 7.0 seconds

So (again ignoring everything else) the printer will take 6+6.5+7=19.5 seconds to print one layer of each test. Each slice in the output file includes pixels for only one of those tests, but of course the pixels never overlap.

Would it therefore be possible to generate the slices as follows:

Layer 1: A + B + C, 6.0 seconds
Layer 2: B + C, 0.5 seconds
Layer 3: C, 0.5 seconds

The total print time in that case would only be 7 seconds per layer, while preserving the correct exposure times.

## Describe alternatives you've considered

I'm still fairly new to UVtools (which, btw, thanks for such a useful and versatile suite of tools!), so I have a hunch that merging layers and retiming them in this way might be possible via scripting. At present, though, it's outside of my ability.

## Additional context

I'm far from an expert in the mechanics of resin printing, so if there's some technical reason why this is impossible, please forgive my ignorance. :)

sn4k3, 2021-04-12:
That will invalidate the test.
Cure a layer for 6s + 0.5s is not the same as cure in once 6.5s layer.
That would only be possible if LCD could render the image almost instant.
Right now with obligatory lifts, firmware restrictions, light offs etc, the layer will cooldown on that time and them recure for other time, you you do that you will find the test will not match to a real print where layer cure for a continous time.
Even if you have a gcode printer = no lifts, no lightoffs, change layer image takes time and layer will cooldown.
Also its bad ideia to keep LED ON almost all the time, lifts, lightoff provide a cooldown to the LCD which extend your hardware lifetime.

sn4k3, 2021-08-01:
Already optimized in the best way with recent releases

closed 2021-08-01 by sn4k3