sn4k3/UVtools #186: [FEATURE REQUEST] Cursor coordinates

MartinLunn, 2021-04-23:
## Is your feature request related to a problem? Please describe

Background:
Workflow:
1. Manual support in chitubox
2. Slice, and save to file
3. Open with UVTools and detect problems
4. For fixing islands, I try to locate where the island is on the 3d model to determine if fixing the island would adjust the part's look or function before making any changes. This is useful to me because if the island is on a support structure I can be very heavy handed, or if it's on a detailed and smooth section like a face, I want to be as subtle as I can be.

Problem:
When I click on an automatically detected issue from the list:

[image]

It brings me to a zoomed in view so that I can see the problem. This is great.

Actual Problem:
I find it very frustrating when I'm trying to locate where that issue is once I zoom out. Further, when there are multiple issues very close together, it's difficult to determine which one is being detected.

## Describe the solution you'd like
1. Provide current coordinates in the 2d view of the cursor

OR

2. Have an option to highlight the current issue

or both :)

Possible Location:

[image]

## Describe alternatives you've considered
It's really frustrating to have to maintain my mouse on the issue while zooming out. If I zoom out too much, the "zoom out while centred on the cursor" feature doesn't work.

sn4k3, 2021-04-23:
That resin trap example is bad, as they are so small and are probably fake traps.
Selected issues have a slight different color which you can configure, but the solution to your problem is: 
[image]

You can also disable this:
[image]
OR hold ALT while you select a issue, that will invert that setting, which will center on issue without zoom

MartinLunn, 2021-04-23:
Thanks for the awesome reply, I'll give it a shot!

closed 2021-04-30 by sn4k3